myhomeiot/esphome-components #1: Don't working on ESPHome v2021.9.0, need downgrade

AlexQuaken, 2021-09-16:
Don't working on ESPHome v2021.9.0, need downgrade

myhomeiot, 2021-09-16:
Thanks for information, I already aware about this problem and looking for a fix which will work for 2021.9 and earlier versions.

myhomeiot, 2021-09-17:
I think [this](https://github.com/esphome/esphome/pull/1627/files#diff-e37fc1bf39840409676f55acfd3c7c04f1fe6be6f1852bcbf8b1e4ce61229eb5R73) change the source of the problem, `as_hex_array` was replaced by `as_reversed_hex_array`, not sure how the regular ESPHome BLE Client will works with it, but will see in future releases.

The problem was fixed in [1.0.1](https://github.com/myhomeiot/esphome-components/releases/tag/1.0.1) please update and let's me know if everything works as expected.
Thanks in advance.

AlexQuaken, 2021-09-17:
Still not working ((
see attached logs

[log.txt](https://github.com/myhomeiot/esphome-components/files/7188532/log.txt)

myhomeiot, 2021-09-18:
Log file shows that it's can't connect, sometimes it's happened, let it working for few hours and see.

myhomeiot, 2021-10-02:
If your case required (device tracking, etc) to specify too small [`scan_parameters -> duration`](https://esphome.io/components/esp32_ble_tracker.html) and in the log you will see "ESP-IDF BLE scan never terminated, rebooting to restore BLE stack..." slightly increase duration until ESP stop restarting.

closed 2021-10-02 by myhomeiot