myhomeiot/esphome-components #2: battery sensor publish to ble monitor

marcelveldt, 2021-12-09:
Hi, great work on this, exactly what I was looking for!

One question though. Is there any modification needed on the HA ble_monitor component to get the battery values ?
I used your example to publish them to the parse service but I don't see the battery sensor appearing in HA.

myhomeiot, 2021-12-10:
@marcelveldt Thanks ;)

Yes, you need to modify ble_monitor, I add comment [here](https://github.com/myhomeiot/esphome-components/blob/main/examples/myhomeiot_ble_client/flower_care_battery_ble_advertise.yaml).
Currently ble_monitor can't create sensor when receives data so we can't ask to include this changes into ble_monitor because all other users will get battery sensor with unknown/unavailable state.

marcelveldt, 2021-12-10:
Thanks, that was the missing piece of the puzzle.
Maybe the HA component should create this sensor as soon as the first value is received. That way it will work for all users.
If I have some spare time next week I'll see if I can PR that.

closed 2021-12-10 by marcelveldt

myhomeiot, 2021-12-17:
@marcelveldt Battery sensor was added for all users with note in Passive BLE Monitor [6.3.0](https://github.com/custom-components/ble_monitor/releases/tag/6.3.0)

marcelveldt, 2021-12-17:
Yeah, I spotted that. I think this could be a cool project for creating a house wide sensor setup.
Just have some bluetooth dongles and/or some ESPHome devices throughout the device and you have covered your whole house.

marcelveldt, 2021-12-17:
BTW: Passing the devices from HA to the ESP is also possible with a service call:

```
api:
  services:
    - service: set_devices
      variables:
        mac_addresses: string
      then:
        lambda: id(blegateway).set_devices(x);
```

I'm currently looking at the ble_monitor code to see if I can add some logic to auto detect the ESP's running the ble gateway and pass the devices list. That would be super cool and even more easy.
Populate a list in BLE Monitor with all found backends (ESP's and/or USB dongles).

marcelveldt, 2021-12-17:
@chemelli74 nice to see you here too :-)