myhomeiot/esphome-components #3: Battery sensor always 0%

chemelli74, 2021-12-10:
I implemented your code for firmware and battery.

Only changed 2 things:
- decimal: 1 (for battery)
- icon: mdi:chip (for firmware)

This is the result:

[image]

How can I debug the issue of battery sensor ?

P.S.> From battery point of view, is better to go with "flower_care_battery_ble_advertise.yaml" than "flower_care.yaml" like I did ?

Thx in advance,

Simone

myhomeiot, 2021-12-10:
It's strange, I also tested it at firmware **3.3.1**, it's should work properly. Can you check the battery level with **Flower Care** application, if it's also shows 0% try to replace the battery.

Decimal 1 for battery of Mi Flora it's a pointless because battery stores in one byte and doesn't have decimal point.

From battery point of view "flower_care_battery_ble_advertise.yaml" and "flower_care.yaml" is equal because it's use same method, but you can do it less frequently by specifying update_interval, in examples its 24h which I think is reasonable.

If you use [Passive BLE Monitor](https://github.com/custom-components/ble_monitor) for sensors, better to use "flower_care_battery_ble_advertise.yaml" and have all data in one place (BLE Monitor) instead ESP.
You can also use [BLE Gateway](https://github.com/myhomeiot/esphome-components#ble-gateway) which makes ESP as BLE repeater and with BLE Gateway you don't need to have other components which parses data from devices at ESP, BLE packages will be sent and parse at Passive BLE Monitor.

chemelli74, 2021-12-11:
> It's strange, I also tested it at firmware **3.3.1**, it's should work properly. Can you check the battery level with **Flower Care** application, if it's also shows 0% try to replace the battery.

Seems that under 2%-3% in the app, I get 0% from this component.

> Decimal 1 for battery of Mi Flora it's a pointless because battery stores in one byte and doesn't have decimal point.

Thx, updated.

 
> From battery point of view "flower_care_battery_ble_advertise.yaml" and "flower_care.yaml" is equal because it's use same method, but you can do it less frequently by specifying update_interval, in examples its 24h which I think is reasonable.

I left the 24h as per your sample.

> If you use [Passive BLE Monitor](https://github.com/custom-components/ble_monitor) for sensors, better to use "flower_care_battery_ble_advertise.yaml" and have all data in one place (BLE Monitor) [...]

I don't have bluetooth on HA, it's running as a docker container in my NAS.

Simone

myhomeiot, 2021-12-11:
I did some tests with Flower Care app, in my case (IOS). I manage to connect to MiFlore with old batteries (less than 20%) after 10-15 retries, and I get --%, 0% and some > 10% < 20% but never seen 2-3% maybe application did some adjustment’s if battery too low. In this tests component show values > 0.
Try to play with different batteries and check reported values, did you get something > 0, or component always returns 0%?

BTW the firmware 3.3.5 is available from app, I did update and it's looks fine (I suggest to use new battery for firmware update, after update you can install old one).

You can use Passive BLE Monitor at the host without any BT adapters, and configure it in [yaml](https://github.com/myhomeiot/esphome-components#home-assistant-passive-ble-monitor-configuration-example) without specifying bt_interface. BLE Gateway from ESP will feed Passive BLE Monitor with data.

myhomeiot, 2021-12-12:
Finally I get 0% from component, when I connect to device with Flower Care app it's also shows 0%. Will see how this will changes.

chemelli74, 2021-12-16:
> I did some tests with Flower Care app, in my case (IOS). I manage to connect to MiFlore with old batteries (less than 20%) after 10-15 retries, and I get --%, 0% and some > 10% < 20% but never seen [...]

Played with different batteries and finally got values > 0.

> BTW the firmware 3.3.5 is available from app, I did update and it's looks fine (I suggest to use new battery for firmware update, after update you can install old one).

I'm not offered this version when I go in the app menu: still says 3.3.1 is the latest :-S

> You can use Passive BLE Monitor at the host without any BT adapters, and configure it in [yaml](https://github.com/myhomeiot/esphome-components#home-assistant-passive-ble-monitor-configuration-example) without specifying bt_interface. BLE Gateway from ESP will feed Passive BLE Monitor with data.

I will consider switching then, seems make all things easier to handle

Simone

ersate, 2021-12-17:
@myhomeiot hi, what is the proper and preffered method of contacting you ?

myhomeiot, 2021-12-17:
> @myhomeiot hi, what is the proper and preffered method of contacting you ?

I created a thread [here](https://community.home-assistant.io/t/esphome-ble-gateway-and-other-ble-components) where you can post your question or send PM.
Also added this link to README, hope this helps for other users.

myhomeiot, 2021-12-27:
> Finally I get 0% from component, when I connect to device with Flower Care app it's also shows 0%. Will see how this will changes.

@chemelli74 It's still shows 0% battery to FlowerCare application and keep working. It's looks like problem of battery percentage calculation on device, and we can't do anything. If 0% is annoying, you can add if to the [lambda](https://github.com/myhomeiot/esphome-components/blob/main/examples/myhomeiot_ble_client/flower_care_battery_ble_advertise.yaml) and check if it's report value less than 10% count it as 10%.

closed 2021-12-27 by myhomeiot

myhomeiot, 2022-01-15:
@chemelli74 After few weeks, device which shown 0% start report battery level 13%. I think for this devices we can use battery level just for information and replace battery only after device became unavailable.

chemelli74, 2022-01-15:
@myhomeiot, I also think that each restart of HA fires a new request for fw and battery %, thus consuming a lot of battery.
This is true also for the other passive values, that shows as unavailable upon HA restart. LEss cretical as they don't consume battery but still not nice to see.

Is there a way for esphome to cache those data and avoid requesting them each time ?

Simone

myhomeiot, 2022-01-15:
@chemelli74, You are sure? The new fw and battery request can be fired only if ESP is restarted or `update_interval` of sensor is reached, by default ESP restarts if during [15min](https://esphome.io/components/api.html) it's doesn't have connection from HA, but you can change it ESPHome config:

```
api:
  reboot_timeout: 1h
```

Also check that you have `restore_state` at Passive BLE Monitor configuration, it's will preserves sensors values during HA restarts.
```
ble_monitor:
  ...
  restore_state: true
```

myhomeiot, 2022-07-08:
I have two MiFlora devices which for last 6 months shows 0% but keep working and sending other data as BLE advertisement. I think battery level information from MiFlora is useless and better to check if it's alive using last change timestamp of illuminance reported by device.