myhomeiot/esphome-components #9: Example of Direct Service Call

efaden, 2022-04-09:
Is there an example of using a direct HA service call instead of an automation?

Would it be?  Is there an advantage to one or the other?

```
on_ble_advertise:
  then:
    homeassistant.service:
      event: ble_monitor.parse_data
      data:
        packet: !lambda return packet;
```

myhomeiot, 2022-04-10:
Example of service call you can find in ESPHome documentation [here](https://esphome.io/components/api.html#homeassistant-service-action)

```
  on_ble_advertise:
    then:
      homeassistant.service:
        service: ble_monitor.parse_data
        data:
          packet: !lambda return packet;
```

If you are using direct service call it's make call faster but you need to recompile ESP firmware if, for example you need to temporarily turn processing of BLE Advertisement.

If you are using event and automation it's make call slower but it's more flexible, you can disable automation for example in order to stop processing BLE Advertisement from all ESP devices or make condition and allow processing only from specific devices, add packet counter, etc.

closed 2022-04-14 by myhomeiot

efaden, 2022-10-11:
Got it. Thanks!