myhomeiot/esphome-components #10: If discovery disabled on_ble_advertise not working

DunklerPhoenix, 2022-05-12:
Heho.
I'm trying to get the ble_gateway to work, but if discovery is disabled (which is default) the on_ble_advertise doesn't trigger.

How to enable discovery by default? Without it the examples in you documentation do nothing.

```
esphome:
  name: esphome-kueche-ble-gateway:
  
external_components:
  - source: github://myhomeiot/esphome-components
    
esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ""

wifi:
  ssid: !secret wifi24_username
  password: !secret wifi24_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Kueche-Ble-Gateway"
    password: ""

captive_portal:
    
esp32_ble_tracker:

# ESPHome
ble_gateway:
  id: blegateway
  on_ble_advertise:
    then:
      homeassistant.event:
        event: esphome.on_ble_advertise
        data:
          packet: !lambda return packet;
          gateway_id: ${device_id}

binary_sensor:
  - platform: homeassistant
    id: ble_gateway_discovery
    entity_id: binary_sensor.ble_gateway
    attribute: discovery
    on_state:
      then:
        lambda: id(blegateway).set_discovery(x);

text_sensor:
  - platform: homeassistant
    id: ble_gateway_devices
    entity_id: binary_sensor.ble_gateway
    attribute: devices
    on_value:
      then:
        lambda: id(blegateway).set_devices(x);

switch:
  - platform: template
    id: switch_ble_gateway_discovery
    name: BLE Gateway Discovery
    icon: mdi:bluetooth-connect
    lambda: return id(blegateway).get_discovery();
    turn_on_action: [lambda: id(blegateway).set_discovery(true);]
    turn_off_action: [lambda: id(blegateway).set_discovery(false);]
    disabled_by_default: true
```

DunklerPhoenix, 2022-05-12:
For e.g. this one does nothing:

```
esphome:
  name: esphome-kueche-ble-gateway
  
      
external_components:
  - source: github://myhomeiot/esphome-components
    
esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ""

wifi:
  ssid: !secret wifi24_username
  password: !secret wifi24_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Kueche-Ble-Gateway"
    password: ""

captive_portal:
    
esp32_ble_tracker:

# ESPHome
ble_gateway:
  id: blegateway
  on_ble_advertise:
    then:
      homeassistant.event:
        event: esphome.on_ble_advertise
        data:
          packet: !lambda return packet;
          gateway_id: ${device_id}
```

myhomeiot, 2022-05-13:
Hello,

Discovery it's a configuration parameter and can be enabled as any other parameters in ESPHome (`discovery: true`):
```
ble_gateway:
  id: blegateway
  discovery: true
  on_ble_advertise:
    then:
      homeassistant.event:
        event: esphome.on_ble_advertise
        data:
          packet: !lambda return packet;
          gateway_id: ${device_id}
```
Without discovery you should specify `devices` MAC addresses like in this [example](https://github.com/myhomeiot/esphome-components#esphome-configuration-example) or use [advanced configuration](https://github.com/myhomeiot/esphome-components#advanced-configuration-where-esphome-devices-gets-mac-addresses-from-passive-ble-monitor-configuration) with one of the described methods for initial device detection by Passive BLE Monitor (check important note section of advanced configuration).

DunklerPhoenix, 2022-05-13:
Ahhhh. Thank you. Didn't understand it correctly :P

closed 2022-05-13 by DunklerPhoenix