myhomeiot/esphome-components #12: ValueError: invalid literal for int() with base 10: '0-dev' when compiling on esphome 2022.7.0-dev

mihsu81, 2022-07-05:
Hi @myhomeiot,

Thanks a lot for the awesome job. 😊👍

When compiling on esphome 2022.7.0-dev for a Lolin C3 Mini it throws an error: _ValueError: invalid literal for int() with base 10: '0-dev'_. Do you have any suggestions to make it work?
Thank you in advance.

```
INFO Reading configuration /config/esphome/ble-gateway04.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Bucharest'
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 931, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 918, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 330, in command_compile
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 163, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 175, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 613, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 155, in wrapped
    await coro(conf)
  File "/config/esphome/.esphome/external_components/4e725a3b/components/myhomeiot_ble_client/__init__.py", line 57, in to_code
    reversed = versiontuple(const.__version__) >= versiontuple("2021.9.0")
  File "/config/esphome/.esphome/external_components/4e725a3b/components/myhomeiot_ble_client/__init__.py", line 52, in versiontuple
    return tuple(map(int, (v.split("."))))
ValueError: invalid literal for int() with base 10: '0-dev'
```

```
esp32:
  board: lolin_c3_mini
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0
```

myhomeiot, 2022-07-05:
Try with latest fix.

mihsu81, 2022-07-05:
Still getting the same error. I've already cleaned build files. Do I need to do anything else?

```
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 931, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 918, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 370, in command_run
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 163, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 175, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 613, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 155, in wrapped
    await coro(conf)
  File "/config/esphome/.esphome/external_components/4e725a3b/components/myhomeiot_ble_client/__init__.py", line 57, in to_code
    reversed = versiontuple(const.__version__) >= versiontuple("2021.9.0")
  File "/config/esphome/.esphome/external_components/4e725a3b/components/myhomeiot_ble_client/__init__.py", line 52, in versiontuple
    return tuple(map(int, (v.split("."))))
ValueError: invalid literal for int() with base 10: '0-dev'
```

closed 2022-07-05 by mihsu81