myhomeiot/esphome-components #15: ble_gateway: homeassistant.service from esphoe automation

tbaumann, 2022-08-11:
Is there any reason why this shouldn't work?
```yaml
  on_ble_advertise:
    then:
    - logger.log: BLE  !lambda return packet;
    - homeassistant.service:
        service: ble_monitor.parse_data
        data:
          packet: !lambda return packet;
          gateway_id: $device_name
```

It seems a bit unnecessary to rely on a homeassistant automation to listen to the event first

tbaumann, 2022-08-11:
Actually, it might work. I think I had other issues why my device didn't show up in HA.

So perhaps this is just a documentation suggestion. This seems just so much more straight forward to implement.

myhomeiot, 2022-08-11:
Should work, let's me know if you found this broken.

Main advantage to have automation instead service call it's you can disable automation if you want to stop ESP to send BLE packages.

tbaumann, 2022-08-11:
Thanks. Glad to see this is okay. I will debug my issue a little more. But I don't think it was related.

I'm always a fan of simple solutions. But I agree, the event has the benefit of leaving an obvious debug trace in HA and have possible conditionals attached.

closed 2022-08-11 by tbaumann