myhomeiot/esphome-components #16: ble_gateway: Document discovery mode

tbaumann, 2022-08-11:
I just came across the option `discovery:  true`

I think this is pretty much the default behavior anyone would expect of a BLE gateway. And it's a lot more easy to understand than the currently documented method with the text_sensor.

I think this option should be mentioned in the documentation and examples.

My minimal config is this. 
```yaml
ble_gateway:
  id: $device_name
  discovery: true
  on_ble_advertise:
    then:
    - homeassistant.service:
        service: ble_monitor.parse_data
        data:
          packet: !lambda return packet;
          gateway_id: $device_name
```

Add this to any esphome device and receive BLE advertisements. Simple.

myhomeiot, 2022-08-11:
I will think how to add this into documentation to make things clear.

Main disadvantage of automatic discovery it's if you are living in environment with many BLE devices around it's can overload Passive BLE Monitor. Earlyer versions of Passive BLE Monitor in my case takes 50-60% CPU (instead 5-7%) for packet processing when ESP sends all BLE advertaisement packets.

tbaumann, 2022-08-11:
Oh the load issues are surprising. Passive BLE Monitor, when using a local HCI interface also default to listening to everything AFAIK.

Hard to say what's the better default I guess.

myhomeiot, 2022-08-11:
Yes, just found that now Passive BLE Monitor listening and parsing all packets because auto [discovery](https://custom-components.github.io/ble_monitor/configuration_params#discovery) enabled by default, probably the problem with heavy CPU load was solved.