myhomeiot/esphome-components #17: ble_gateway: Raw packet to MQTT

JacoBezuidenhout, 2022-08-22:
Hi everyone. 

Wow this is really cool! I am fairly new to ESPHome and am struggling to publish the raw packet to mqtt (and not home-assistant). How will I go about it?

myhomeiot, 2022-08-22:
I will give you example soon :smile:

closed 2022-08-22 by myhomeiot

myhomeiot, 2022-08-22:
```
mqtt:
  id: mqtt_client
  broker: 192.168.1.10
  username: mqtt_username
  password: mqtt_password
  reboot_timeout: 1h

ble_gateway:
  id: blegateway
  on_ble_advertise:
    then:
      mqtt.publish:
        topic: ble_gateway
        payload: !lambda return packet;
```

More info about ESPHome MQTT you can find here:
- [MQTT Client Component](https://esphome.io/components/mqtt.html)
- [mqtt.publish Action](https://esphome.io/components/mqtt.html#mqtt-publish-action)

JacoBezuidenhout, 2022-08-22:
Thank you verrrrryyy much!! I overcomplicated it completely. I really appreciate the super quick response!

JacoBezuidenhout, 2022-08-22:
Just a note to other noobs like myself:

I had to add discovery: true to ble_gateway for it to start steaming over MQTT.

mrmaximas, 2025-07-18:
@myhomeiot Hi! Thanks for this project.
Can you help me convert raw packets from Sensirion STH4x Smart Gadget like these to mqtt topics?
```
[17:46:01][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B2F645E840D09534854343020476164676574A9
[17:46:31][D][esp32_ble_tracker:343]: End of scan, set scanner state to IDLE.
[17:46:31][D][esp32_ble_tracker:307]: Starting scan, set scanner state to STARTING.
[17:46:43][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B3C641E840D09534854343020476164676574B6
[17:47:13][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B3B64FD830D09534854343020476164676574B4
[17:47:21][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B346424840D09534854343020476164676574AB
[17:47:52][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B326443840D09534854343020476164676574BB
[17:47:59][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B326443840D09534854343020476164676574B4
[17:49:38][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B3D64AC840D09534854343020476164676574A9
[17:50:38][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B3F6494840D09534854343020476164676574B6
[17:51:16][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B3E6464840D09534854343020476164676574B7
[17:51:31][D][esp32_ble_tracker:343]: End of scan, set scanner state to IDLE.
[17:51:31][D][esp32_ble_tracker:307]: Starting scan, set scanner state to STARTING.
[17:52:28][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B426483840D09534854343020476164676574A6
[17:52:58][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B42647A840D09534854343020476164676574B7
[17:53:28][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B3F6495840D09534854343020476164676574B6
[17:54:06][D][ble_gateway:071]: [FF:32:88:71:E5:8B] Packet 043E29020100018BE5718832FF1D0201060BFFD5060006E58B446459840D09534854343020476164676574BA
```
Oddly enough they are 44 bytes each, not 31 as it says here.
https://github.com/akx/sensirion-ble/blob/main/reference/Sensirion_BLE_communication_protocol.pdf
Where byte 26 is higher byte of temperature value, byte 25 lower byte of temperature value, device MAC address FF:32:88:71:E5:8B is seen in reverse order, device ID (las two bytes of MAC) in normal order (bytes 23-24)
The actual temperature tick bytes are as follows:
2F64
3C64
3B64
3464
3264
3264
3D64
For examle for firts ticks
0x64*256+0x2F=100*256+47 =25647 
temp = ((175 * temp_ticks) / 65535) -45 = 23.48 degrees.
But! I don't know how to realise it with esphome :(

myhomeiot, 2025-07-21:
As you already [familiar](https://github.com/myhomeiot/esphome-components/issues/37#issuecomment-3088877620) with publishing data into template sensors you can use same for parsing Sensirion STH4x Smart Gadget data.
It's doesn't required `ble_gateway` component, you can use `esp32_ble_tracker` event trigger 'on_ble_advertise'. Sure the logic should be more complicated and in this example you should change indexes of scan_result.ble_adv[1-2] for real indexes after you see what ESP_LOGI prints for your sensor (ble_gateway adds extra data for ble_monitor).
Passive BLE Monitor already has support of your sensor so you can get logic from it [here](https://github.com/custom-components/ble_monitor/blob/018cfca91c398ed134df1a8b3ad639c18caa9a53/custom_components/ble_monitor/ble_parser/sensirion.py#L191)

Hope this helps.

```yaml
esp32_ble_tracker:
  on_ble_advertise:
    - mac_address:
        - FF:32:88:71:E5:8B
      then:
        - lambda: |-
            auto scan_result = x.get_scan_result();
            ESP_LOGI("on_ble_advertise", "%s: %s", x.address_str().c_str(), format_hex_pretty(scan_result.ble_adv, scan_result.adv_data_len + scan_result.scan_rsp_len).c_str());
            float temp_ticks = scan_result.ble_adv[2] * 256 + scan_result.ble_adv[1];
            float temp = ((175 * temp_ticks) / 65535) - 45;
            id(sensirion_temperature).publish_state(temp);

sensor:
  - platform: template
    id: sensirion_temperature
    name: Sensirion Temperature
    device_class: temperature
    accuracy_decimals: 2
```

mrmaximas, 2025-07-21:
@myhomeiot today i solve this by myself too ;)
#  SHT4x Smart Gadget
  on_ble_manufacturer_data_advertise:
    - mac_address: FF:32:88:71:E5:8B
      manufacturer_id: 06d5
      then:
        - lambda: |-
            float t = (((x[4] + (x[5] << 8)) * 175.0)/65535 - 45.0);
            id(sht4x_t).publish_state(t);
            float h = (((x[6] + (x[7] << 8)) * 125.0)/65535 - 6.0);
            id(sht4x_h).publish_state(h);

sensor:

# SHT45 Smart Gadget

  - platform: template
    id: sht4x_t     
    name: "SHT4x Temperature"
    accuracy_decimals: 2

  - platform: template
    id: sht4x_h      
    name: "SHT4x Humidity"
    accuracy_decimals: 2