myhomeiot/esphome-components #20: Identifying service/characteristics uuid

bipsendk, 2022-12-30:
How does one identify the uuids to be used ?

In a config with the "native" ble_client, I currently have this in the sensor section:

```
ble_client:
  - mac_address: FF:D1:01:5E:87:70
    id: Oras6160FZ_Bathroom

sensor:
  - platform: ble_client
    type: characteristic
    ble_client_id: Oras6160FZ_Bathroom
    name: "oras_ValveCounter"
    service_uuid: "2BE32DB1-5F6B-4CBD-8803-38D6DFB16490"
    characteristic_uuid: "2BE32DB1-5F6B-4CBD-8873-8D6DFB164900"
    update_interval: 300s
    notify: false
    lambda: |-
      uint8_t* pdata = (uint8_t*) x.data();
      unsigned long int totConsumption = pdata[0];
      totConsumption += ( pdata[1]<<8);
      totConsumption += ( pdata[2]<<16);
      totConsumption += ( pdata[3]<<24);
      id(oras_totConsumption).publish_state(totConsumption);
      id(oras_totConsumption_m3).publish_state((float)totConsumption/(float)1000);
      ESP_LOGI("ble_adv", "Total water consumption %ld", totConsumption);
      unsigned long int totOpening = pdata[4];
      totOpening += ( pdata[5]<<8);
      totOpening += ( pdata[6]<<16);
      totOpening += ( pdata[7]<<24);
      ESP_LOGI("ble_adv", "Total valve openings %ld", totOpening);
      ESP_LOGI("ble_adv", "    - Setting bleStatus to true");
      id(bleStatus).publish_state(true);
      return (float)totOpening;
    icon: 'mdi:faucet'
    filters:
      - filter_out: nan
```

What would be the correct way to rewrite this to use myhomeiot_ble_client ?

myhomeiot, 2023-01-03:
Hello,

If you need to query multiple services/characteristics you should write extra independent item of myhomeiot_ble_client with different services/characteristics.
As you can read in myhomeiot_ble_client readme, the component is designed to query services/characteristics and disconnect from device, so every time if update_interval expired it's looking for advertisement from device, connect to it, read characteristic and disconnect from it, you can see this process in the ESPHome log.

Some examples of usage of component you can find [here](https://github.com/myhomeiot/esphome-components/tree/main/examples/myhomeiot_ble_client).
Let's me know if it's helps.

In your case usage of myhomeiot_ble_client should looks something like this:
```yaml
esp32_ble_tracker:

myhomeiot_ble_host:

myhomeiot_ble_client:
  - mac_address: FF:D1:01:5E:87:70
    service_uuid: '2BE32DB1-5F6B-4CBD-8803-38D6DFB16490'
    characteristic_uuid: '2BE32DB1-5F6B-4CBD-8873-8D6DFB164900'
    update_interval: 300s
    on_value:
      then:
        lambda: |-
          unsigned long int totConsumption = x[0];
          totConsumption += x[1] << 8;
          totConsumption += x[2] << 16;
          totConsumption += x[3] << 24;
          //id(oras_totConsumption).publish_state(totConsumption);
          id(oras_totConsumption_m3).publish_state((float)totConsumption / 1000.0f);
          ESP_LOGI("ble_adv", "Total water consumption %ld", totConsumption);
          //unsigned long int totOpening = x[4];
          //totOpening += x[5] << 8;
          //totOpening += x[6] << 16;
          //totOpening += x[7] << 24;
          //ESP_LOGI("ble_adv", "Total valve openings %ld", totOpening);
          //ESP_LOGI("ble_adv", "    - Setting bleStatus to true");
          //id(bleStatus).publish_state(true);
          //id(oras_totOpening).publish_state(totOpening);

sensor:
  - platform: template
    id: oras_totConsumption_m3
    name: Oras Total Consumption
    device_class: water
    unit_of_measurement: 'm³'
    state_class: measurement
    accuracy_decimals: 0
```

mihsu81, 2023-01-11:
@bipsendk Did you manage to get it working reliably?
I have 2 HANSAFIT ECO+ (65412219) which are basically the rebranded Oras Optima ECO+ (1714FZ) and it takes at least 3-4 connection retries before my gateway can get a reading.
Do you have a decoding sample for the other characteristics? 
@myhomeiot is there any way we can save the values to flash to prevent Unknown values after a reboot?

Thanks in advance.

bipsendk, 2023-01-11:
@mihsu81 - I do not have a working solution yet. I am still trying to figure out if it can be done using the myhomeiot BLE client - or I should go with an alternative and scan for the faucet, and fetch the desired data once connected - and then go to deep sleep for 1-2 hours before wakeup and doing it all over again (in order not to drain the battery of the faucet).

The code can probably store data in Flash (simulated drive), but there is a limit on the number of times you can rewrite a section of the flash. This is something to be done in the lambda code, and is not a part of the BLE client code.
An alternative is to attach an external eeprom to the ESP Device (which can handle more rewrites) - or in case of a ESP device reset, simply use template sensors in HA, so you filter out unavailable and unknown values

mihsu81, 2023-01-11:
@bipsendk Thanks a lot for the info. I was asking about persistent values across reboots because I've noticed it possible in dentra's integration. https://github.com/dentra/esphome-components/tree/master/components/miot_cwbs01

bipsendk, 2023-01-13:
@myhomeiot  - when having multiple myhomeiot_ble_client entries towards the same MAC and service uuid, but different characteristics - will all entries be processed "more or less in sequential order" ? Or will the system take them one after the other, more or less using the next BLE advertise as a sort of index counter?

bipsendk, 2023-01-15:
@myhomeiot - just tried to create a config with multiple entries on different characteristics:

```
myhomeiot_ble_client:
  - mac_address: EC:E0:26:98:92:79 # 
    id: Oras6160FZ_Bathroom_8843
    update_interval: never
    service_uuid: '2BE32DB1-5F6B-4CBD-8803-38D6DFB16490'
    characteristic_uuid: '2BE32DB1-5F6B-4CBD-8843-8D6DFB164900'
    on_value:
        then:
            lambda: |-
              uint8_t* pdata = (uint8_t*) x.data();
              uint16_t batteryVoltage = pdata[4];
              uint8_t batteryPct = pdata[6];
              batteryVoltage += ( pdata[5]<<8);

              ESP_LOGD("ble_adv_8843", "BatteryVoltage %f", (float)(batteryVoltage)/100);
              ESP_LOGD("ble_adv_8843", "BatteryPercentage %d", batteryPct);

  - mac_address: EC:E0:26:98:92:79 # 
    id: Oras6160FZ_Bathroom_8853
    update_interval: never
    service_uuid: '2BE32DB1-5F6B-4CBD-8803-38D6DFB16490'
    characteristic_uuid: '2BE32DB1-5F6B-4CBD-8853-8D6DFB164900'
    on_value:
        then:
            lambda: |-
              uint8_t* pdata = (uint8_t*) x.data();
              unsigned long int secsSinceLastReset = pdata[0];
              secsSinceLastReset += ( pdata[1]<<8);
              secsSinceLastReset += ( pdata[2]<<16);
              secsSinceLastReset += ( pdata[3]<<24);
              ESP_LOGD("ble_adv_8853", "Seconds since last reset %ld", secsSinceLastReset);

```

The problem here is, that the clients are newer connected...

I have:

```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

```

And from that output (triggered when an advertie is received) I get:

```
[20:08:04][D][ble_adv:054]: New BLE device
[20:08:04][D][ble_adv:055]:   address: EC:E0:26:98:92:79
[20:08:04][D][ble_adv:056]:   name: ORAS
[20:08:04][D][ble_adv:057]:   Advertised service UUIDs:
[20:08:04][D][ble_adv:061]:   Advertised service data:
[20:08:04][D][ble_adv:065]:   Advertised manufacturer data:
[20:08:04][D][ble_adv:067]:     - 0x0131: (length 18)
[20:08:04][D][ble_adv:080]:   address: EC:E0:26:98:92:79
[20:08:04][D][ble_adv:081]:   name: ORAS
[20:08:04][D][ble_adv:082]:   Advertised service UUIDs:
[20:08:04][D][ble_adv:086]:   Advertised service data:
[20:08:04][D][ble_adv:090]:   Advertised manufacturer data:
[20:08:04][D][ble_adv:092]:     - 0x0131: (length 18)
[20:08:04][I][ble_adv:095]:     - 0x0131: (length 18)
[20:08:04][I][ble_adv:105]: Data    - 00:64:00:32:31:31:31:30:30:37:39:33:35:00:20:20:20:20
[20:08:04][D][sensor:127]: 'oras_BatteryStatus': Sending state 100.00000 % with 1 decimals of accuracy
[20:08:04][I][ble_adv:114]: Serial: 2111007935
[20:08:04][D][ble_adv:054]: New BLE device
[20:08:04][D][ble_adv:055]:   address: EC:E0:26:98:92:79
[20:08:04][D][ble_adv:056]:   name: ORAS
[20:08:04][D][ble_adv:057]:   Advertised service UUIDs:
[20:08:04][D][ble_adv:061]:   Advertised service data:
[20:08:04][D][ble_adv:065]:   Advertised manufacturer data:
[20:08:04][D][ble_adv:067]:     - 0x0131: (length 18)
[20:08:04][D][ble_adv:080]:   address: EC:E0:26:98:92:79
[20:08:04][D][ble_adv:081]:   name: ORAS
[20:08:04][D][ble_adv:082]:   Advertised service UUIDs:
[20:08:04][D][ble_adv:086]:   Advertised service data:
[20:08:04][D][ble_adv:090]:   Advertised manufacturer data:
[20:08:04][D][ble_adv:092]:     - 0x0131: (length 18)
[20:08:04][I][ble_adv:095]:     - 0x0131: (length 18)
[20:08:04][I][ble_adv:105]: Data    - 00:64:00:32:31:31:31:30:30:37:39:33:35:00:20:20:20:20
[20:08:04][D][sensor:127]: 'oras_BatteryStatus': Sending state 100.00000 % with 1 decimals of accuracy
[20:08:04][I][ble_adv:114]: Serial: 2111007935
[20:08:06][D][ble_adv:054]: New BLE device
[20:08:06][D][ble_adv:055]:   address: EC:E0:26:98:92:79
[20:08:06][D][ble_adv:056]:   name: ORAS
[20:08:06][D][ble_adv:057]:   Advertised service UUIDs:
[20:08:06][D][ble_adv:061]:   Advertised service data:
[20:08:06][D][ble_adv:065]:   Advertised manufacturer data:
[20:08:06][D][ble_adv:067]:     - 0x0131: (length 18)
[20:08:06][D][ble_adv:080]:   address: EC:E0:26:98:92:79
[20:08:06][D][ble_adv:081]:   name: ORAS
[20:08:06][D][ble_adv:082]:   Advertised service UUIDs:
[20:08:06][D][ble_adv:086]:   Advertised service data:
[20:08:06][D][ble_adv:090]:   Advertised manufacturer data:
[20:08:06][D][ble_adv:092]:     - 0x0131: (length 18)
[20:08:06][I][ble_adv:095]:     - 0x0131: (length 18)
[20:08:06][I][ble_adv:105]: Data    - 00:64:00:32:31:31:31:30:30:37:39:33:35:00:20:20:20:20
[20:08:06][D][sensor:127]: 'oras_BatteryStatus': Sending state 100.00000 % with 1 decimals of accuracy
[20:08:07][D][ble_adv:054]: New BLE device
[20:08:07][D][ble_adv:055]:   address: EC:E0:26:98:92:79
[20:08:07][D][ble_adv:056]:   name: ORAS
[20:08:07][D][ble_adv:057]:   Advertised service UUIDs:
[20:08:07][D][ble_adv:061]:   Advertised service data:
[20:08:07][D][ble_adv:065]:   Advertised manufacturer data:
[20:08:07][D][ble_adv:067]:     - 0x0131: (length 18)
[20:08:07][D][ble_adv:080]:   address: EC:E0:26:98:92:79
[20:08:07][D][ble_adv:081]:   name: ORAS
[20:08:07][D][ble_adv:082]:   Advertised service UUIDs:
[20:08:07][D][ble_adv:086]:   Advertised service data:
[20:08:07][D][ble_adv:090]:   Advertised manufacturer data:
[20:08:07][D][ble_adv:092]:     - 0x0131: (length 18)
[20:08:07][I][ble_adv:095]:     - 0x0131: (length 18)
[20:08:07][I][ble_adv:105]: Data    - 00:64:00:32:31:31:31:30:30:37:39:33:35:00:20:20:20:20
[20:08:07][D][sensor:127]: 'oras_BatteryStatus': Sending state 100.00000 % with 1 decimals of accuracy
[20:08:07][I][ble_adv:114]: Serial: 2111007935

```

But the myhomeiot_ble_client never seems to connect and run the lambda code.... Even though I have checked service and characteristics uuids ...

myhomeiot, 2023-01-16:
Try to remove/comment out the extra settings of `esp32_ble_tracker`, instead this:

```yaml
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true
```

use just this:
```yaml
esp32_ble_tracker:
```

bipsendk, 2023-01-16:
The parameters on the BLE tracker is in order to locate the device quicker - as the device primarily sends advertise when in use, and not when "idle"  ...

myhomeiot, 2023-01-17:
I will try to test `myhomeiot_ble_client` with your config within few days.

bipsendk, 2023-02-27:
any progress with multiple  "clients" towards same target ?

mihsu81, 2023-07-24:
Hi @myhomeiot ,
Did you manage to have a look at multiple clients?
Thanks you.