myhomeiot/esphome-components #21: Error "READ_CHAR_EVT error reading char at handle (33), status (5)" when reading battery value from soocare.toothbrush.x3

mihsu81, 2023-01-12:
Hi @myhomeiot,
When trying to read the value of the battery characteristic of a soocare.toothbrush.x3 I see the below warning and the value is not read: ```[W] | [myhomeiot_ble_client:194] | [EF:A8:A8:06:8C:E6] READ_CHAR_EVT error reading char at handle (33), status (5)```
With nRF Connect I can see the value is _**17%**_. Is **_%_** the cause of the issue, and how can I remove it?

Thank in advance.

<html>
<body>

23:01:25 | [I] | [myhomeiot_ble_client:032] | [EF:A8:A8:06:8C:E6] Connecting
-- | -- | -- | --
23:01:25 | [I] | [myhomeiot_ble_client:087] | [EF:A8:A8:06:8C:E6] Connected successfully, app_id (1)
23:01:25 | [W] | [myhomeiot_ble_client:194] | [EF:A8:A8:06:8C:E6] READ_CHAR_EVT error reading char at handle (33), status (5)
23:01:25 | [I] | [myhomeiot_ble_client:042] | [EF:A8:A8:06:8C:E6] Disconnecting

</body>
</html>

The configuration is:
```yaml
myhomeiot_ble_client:
  - mac_address: EF:A8:A8:06:8C:E6
    service_uuid: '180F'
    characteristic_uuid: '2A19'
    update_interval: 24h
    on_value:
      then:
        lambda: |-
          id(mihai_s_toothbrush_battery).publish_state(x[0]);
```

myhomeiot, 2023-01-13:
Hello, for [ESP_GATTC_READ_CHAR_EVT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gattc.html#_CPPv4N24esp_ble_gattc_cb_param_t25gattc_read_char_evt_param6statusE) `status (5)` means [ESP_GATT_INSUF_AUTHENTICATION](https://github.com/espressif/esp-idf/blob/edd815af2e7d541b25ff3a74c59a7fe0612df42d/components/bt/host/bluedroid/api/include/api/esp_gatt_defs.h#L177), probably this characteristics required pairing.

When you use nRF Connect did you do a pairing with BT device? Check your BT settings and if you have soocare.toothbrush.x3 paired, delete it and try to get this characteristic in nRF Connect without device pairing.

mihsu81, 2023-01-13:
Indeed, I had the toothbrush paired previously. I've tried from another phone and it asked me to pair in order to read the value of the battery characteristic.

mihsu81, 2023-01-16:
Will myhomeiot_ble_client support device pairing? I haven't seen anything related in the documentation.

Thanks.

myhomeiot, 2023-01-17:
AFAIK ESPHome Bluetooth stack unfortunately doesn't support pairing.

mihsu81, 2023-01-17:
That's unfortunate, but i guess not many use cases. Thank you.

mihsu81, 2023-01-21:
Looks like there's a PR waiting to be approved which will allow pairing, and its corresponding documentation.
Should I reopen in case you'd want to implement it?
https://github.com/esphome/esphome/pull/4258
https://github.com/esphome/esphome-docs/pull/2607

myhomeiot, 2023-01-23:
Nice finding! I hope this PR will be merged and will works.
Yes, please reopen this case when this PR will be merged, but before this, please test pairing by regular ble_client and see if it's working for you.

mihsu81, 2023-01-23:
Once the PR gets approved I'll test with ble_client and let you know.
Thank you :).

mihsu81, 2023-01-30:
Because it might be a while before the PR is merged I've tried adding the PR to the configuration of the gateway.
It doesn't seem to make a difference, and it never gets to the authentication part. In fact, when I pair the toothbrush with my phone it never asks for a key.

I did find anther PR https://github.com/esphome/esphome/pull/3320 which i think was already merged in ESPHome which already implements basic BLE auth/security.

I'm kind of stuck at this point. Is there another type of authentication which is missing from ESPHome that might be used by this toothbrush?

```
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 46
[14:44:49][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=46 gattc_if=3
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[14:44:49][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=7 gattc_if=3
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[14:44:49][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=7 gattc_if=3
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[14:44:49][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=7 gattc_if=3
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[14:44:49][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=7 gattc_if=3
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[14:44:49][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=7 gattc_if=3
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[14:44:49][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=7 gattc_if=3
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 7
[14:44:49][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=7 gattc_if=3
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 6
[14:44:49][V][esp32_ble_client:189]: [0] [EF:A8:A8:06:8C:E6] Service UUID: 0x1800
[14:44:49][V][esp32_ble_client:191]: [0] [EF:A8:A8:06:8C:E6]  start_handle: 0x1  end_handle: 0x7
[14:44:49][V][esp32_ble_client:189]: [0] [EF:A8:A8:06:8C:E6] Service UUID: 0x1801
[14:44:49][V][esp32_ble_client:191]: [0] [EF:A8:A8:06:8C:E6]  start_handle: 0x8  end_handle: 0xb
[14:44:49][V][esp32_ble_client:189]: [0] [EF:A8:A8:06:8C:E6] Service UUID: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
[14:44:49][V][esp32_ble_client:191]: [0] [EF:A8:A8:06:8C:E6]  start_handle: 0xc  end_handle: 0x11
[14:44:49][V][esp32_ble_client:189]: [0] [EF:A8:A8:06:8C:E6] Service UUID: 0x180A
[14:44:49][V][esp32_ble_client:191]: [0] [EF:A8:A8:06:8C:E6]  start_handle: 0x12  end_handle: 0x1e
[14:44:49][V][esp32_ble_client:189]: [0] [EF:A8:A8:06:8C:E6] Service UUID: 0x180F
[14:44:49][V][esp32_ble_client:191]: [0] [EF:A8:A8:06:8C:E6]  start_handle: 0x1f  end_handle: 0x22
[14:44:49][V][esp32_ble_client:189]: [0] [EF:A8:A8:06:8C:E6] Service UUID: 00001530-1212-EFDE-1523-785FEABCD123
[14:44:49][V][esp32_ble_client:191]: [0] [EF:A8:A8:06:8C:E6]  start_handle: 0x23  end_handle: 0x2a
[14:44:49][V][esp32_ble_client:189]: [0] [EF:A8:A8:06:8C:E6] Service UUID: 0xFE95
[14:44:49][V][esp32_ble_client:191]: [0] [EF:A8:A8:06:8C:E6]  start_handle: 0x2b  end_handle: 0xffff
[14:44:49][I][esp32_ble_client:193]: [0] [EF:A8:A8:06:8C:E6] Connected
[14:44:49][V][esp32_ble_client:069]: [0] [EF:A8:A8:06:8C:E6]  characteristic 0x2A19, handle 0x21, properties 0x12
[14:44:49][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 18
[14:44:49][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=18 gattc_if=3
[14:44:49][V][esp32_ble_client:157]: [0] [EF:A8:A8:06:8C:E6] cfg_mtu status 0, mtu 23
[14:44:49][V][component:200]: Component esp32_ble took a long time for an operation (0.25 s).
[14:44:49][V][component:201]: Components should block for at most 20-30ms.
[14:44:49][D][esp32_ble_tracker:238]: Starting scan...
[14:44:49][V][esp32_ble:179]: (BLE) gap_event_handler - 2
[14:44:49][V][esp32_ble:179]: (BLE) gap_event_handler - 7
[14:44:50][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:44:51][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:44:51][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:44:51][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:44:52][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:44:52][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:44:52][V][esp32_ble:179]: (BLE) gap_event_handler - 20
...
[14:45:40][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:45:40][V][esp32_ble_tracker:568]: Unhandled type: advType: 0x2b
[14:45:40][V][esp32_ble:179]: (BLE) gap_event_handler - 3
...
[14:46:33][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:46:34][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:46:34][V][esp32_ble:207]: (BLE) gattc_event [esp_gatt_if: 3] - 3
[14:46:34][V][esp32_ble_client:111]: [0] [EF:A8:A8:06:8C:E6] gattc_event_handler: event=3 gattc_if=3
[14:46:34][W][ble_sensor:080]: Error reading char at handle 33, status=5
[14:46:34][V][esp32_ble:179]: (BLE) gap_event_handler - 3
[14:46:34][V][esp32_ble:179]: (BLE) gap_event_handler - 3
```

mihsu81, 2023-07-24:
Hi @myhomeiot,
Did you have a chance to look at simple pairing without authentication?
Thank you.