myhomeiot/esphome-components #22: Delonghi Coffee Machine Example

bubez81, 2023-02-16:
Hi, I noticed your example for the delonghi.
I compiled the code without problems adapting it to my machine and it seems to connect correctly. None of the switches seem to work though:
[image]

[logs_cucina_logs.txt](https://github.com/myhomeiot/esphome-components/files/10759443/logs_cucina_logs.txt)

Could it be a machine setting? Or did something change with a firmware update?
Does it work without problems for you?

myhomeiot, 2023-02-17:
Hello, it's doesn't work with latest versions of ESPHome, something was changed with ble_client text_sensor. Last working version of ESPHome was 2022.11.5.
If you know how to compile it in ESPHome 2022.11.5, you can try.

You can check if it's communicate with machine if in the log you see ```>>>``` requests and ```<<<``` replies.
Sometimes machine BT is stuck, you can try to fully power off machine (unplug the cable) and power on it again. BT module of machine working even it's powered off. If communication is OK, you should also see BT sign on the screen of machine sure if it's exists.

```
[13:13:25][D][coffee_machine:048]: >>> 0D.05.75.0F.DA.25 (6)
[13:13:25][D][text_sensor:067]: 'text_sensor_coffee_machine': Sending state '\xd0'
[13:13:25][D][coffee_machine:162]: D0
[13:13:25][D][text_sensor:067]: 'text_sensor_coffee_machine': Sending state 'u'
[13:13:25][D][coffee_machine:162]: 12.75.0F
[13:13:25][D][text_sensor:067]: 'text_sensor_coffee_machine': Sending state ''
[13:13:25][D][coffee_machine:162]: 01.01.00.00.00.00 (6)
[13:13:25][D][text_sensor:067]: 'text_sensor_coffee_machine': Sending state ''
[13:13:25][D][coffee_machine:162]: 02.00.00.00.00.00 (6)
[13:13:25][D][text_sensor:067]: 'text_sensor_coffee_machine': Sending state ''
[13:13:25][D][coffee_machine:162]: 00.37.4E
[13:13:25][D][coffee_machine:171]: <<< D0.12.75.0F.01.01.00.00.00.00.02.00.00.00.00.00.00.37.4E (19)
```

bubez81, 2023-02-17:
Hello! Yes I can see the BT sign on the screen...let's try to downgrade and see

bubez81, 2023-02-17:
nothing, not even changing the version seems to work:
[image]
I also unplugged the power but nothing to do
not even with the plugin in hacs for homeassistant it works even if it used to work

It's like something has changed lately....

myhomeiot, 2023-02-17:
Which model of Delonghi do you have?

bubez81, 2023-02-17:
The Dinamica Plus
Sure..:
[image]
 I used the hacs plugin for a while https://github.com/Arbuzov/home_assistant_delonghi_primadonna
it worked for months then suddenly stopped working and that's why i tried this system

myhomeiot, 2023-02-17:
This HACS plugin works same way, so probably something was changed in machine side :(

bubez81, 2023-02-17:
Yes, I think so too, I was hoping it was something common to all models but I'm starting to think it's only on my model

bubez81, 2023-02-24:
the very strange thing is that with this https://github.com/mmastrac/longshot the machine works...can't understand

bubez81, 2023-02-24:
ok i think i understand the problem. The machine doesn't accept more than one command simultaneously so if I monitor the status I can't launch the command to make a coffee so I can't connect to the machine continuously but only when I start the command