myhomeiot/esphome-components #25: GL.iNet GL-S10 as BLE Gateway

to4ko, 2023-07-07:
I'm trying  to use GL.iNet GL-S10 as a BLE gateways and no BLE packets received at all. Regular ESP32 board based BLE Gateway working just fine. 

Same time ble device_tracker binary sensor working fine on GL-S10's

`  - platform: ble_presence
    mac_address: 7E:FE:3E:EC:59:FB
    name: ${board_name}_White
    device_class: presence`

Config used for devices https://github.com/to4ko/myconfig/blob/master/esphome/esp32_poe_ble_hs.yaml

BLE Gateway Events counters
[image]

BLE device_tracker's binary Sensors

[image]

myhomeiot, 2023-07-07:
Very strange. Try to enable discovery for this specific device and see if you have any messages from `ble_gateway` in the log. Maybe something wrong with MAC address byte order, etc.

```yaml
ble_gateway:
  id: ${board_name}
  discovery: true
  ...
```

to4ko, 2023-07-07:
no changes with discovery true...

here is log from regular ESP32 ble gateway
[image]

and here is log from GL-S10 device. devices laying next to each other
[image]

myhomeiot, 2023-07-07:
Try to comment out `bluetooth_proxy` configuration and if `ble_gateway` start working return bluetooth_proxy and comment just active: true of it.
PS: I never tested ble_gateway together with bluetooth_proxy :(

```yaml
bluetooth_proxy:
  active: true
```

to4ko, 2023-07-07:
> Try to comment out `bluetooth_proxy` configuration and if `ble_gateway` start working return bluetooth_proxy and comment just active: true of it. PS: I never tested ble_gateway together with bluetooth_proxy :(
> 
> ```yaml
> bluetooth_proxy:
>   active: true
> ```

It's working! Looks like it's not possible to have BLE gateway and BT proxy on the same device...
I'm closing this one as solved!

Thanks!

closed 2023-07-07 by to4ko

myhomeiot, 2023-07-07:
@to4ko Great! Thanks for update!
You can try to place `ble_gateway` before `bluetooth_proxy` in configuration this may works because `ble_gateway` [not marks](https://github.com/myhomeiot/esphome-components/blob/b704eade9be352ac4c55a0140e7d7ae821445cab/components/ble_gateway/ble_gateway.cpp#L66) BLE advertisement as processed and they should pass to next service registered in the chain.

to4ko, 2023-07-07:
test result negative - ble gateway not working once ble proxy enabled in config

myhomeiot, 2023-07-08:
@to4ko This is a [source](https://github.com/esphome/issues/issues/4641) of the problem, in version ESPHome 2023.6.0+ they add **Bluetooth Proxy: Raw bundled advertisements esphome#[4924](https://github.com/esphome/esphome/pull/4924) by @jesserockz**
I know how to fix it but I will wait for ESPHome dev's final decision because this improvement breaks other BLE components which should be modified or dev's should fix it.

myhomeiot, 2023-07-10:
[Issue](https://github.com/esphome/esphome/pull/5073) fixed in ESPHome [2023.6.5](https://github.com/esphome/esphome/releases/tag/2023.6.5).

to4ko, 2023-07-10:
Thanks!!

Will try it!

One question - do you have any expirience with single ble gateway and more than one HA instances?

myhomeiot, 2023-07-10:
> One question - do you have any expirience with single ble gateway and more than one HA instances?

I don't have experience with this, but I think about this before :)
If you have few HA instances probably you have external MQTT, so you can publish `ble_gateway` packets to MQTT and use it in any number of HA instances, more details you can find [here](https://github.com/myhomeiot/esphome-components/issues/17).

to4ko, 2023-07-13:
> I don't have experience with this, but I think about this before :) If you have few HA instances probably you have external MQTT, so you can publish `ble_gateway` packets to MQTT and use it in any [...]

it's working natively without MQTT. Below is a screenshot from second HA instance - parse_data service working on both HAs simultaneously 

[image]

myhomeiot, 2023-07-13:
@to4ko I don't know that you can connect ESPHome to two instances of HA simultaneously. Just seen that in latest version of ESPHome in order to be able to call services you will need [to enable it](https://github.com/home-assistant/core/pull/95143) for new connected ESP.

thematrixdev, 2024-12-06:
Hello @to4ko 
Are you using GL-S10 hardware version v1 or v2?
May you provide the full ESPHome configuration for my reference please?
Thanks.

to4ko, 2024-12-07:
> Hello @to4ko Are you using GL-S10 hardware version v1 or v2? May you provide the full ESPHome configuration for my reference please? Thanks.

hi! please find my config under link below. dont remember actual version :) comparing my config vs [this site](https://devices.esphome.io/devices/GL-iNet-GL-S10) - i have v1

https://github.com/to4ko/myconfig/blob/master/esphome/esp32_poe_ble_hs.yaml