myhomeiot/esphome-components #26: BLE Gateway and BLE Client stopped working after ESPHome update

czesiu20, 2023-11-18:
After update my ESP32 device from 2023.10.6 to 2023.11.2 both ble_gateway and myhomeiot_ble_client stopped working.
The compilation process was successful, without any errors.
The log shows neither received packets nor client connections. It shows nothing... silence...

I'm using the device to read Mi Flora (HHCCJCY01) and Hygro thermometer clock (LYWSD02).
Ble client is used to read the battery status of Mi Flora.

myhomeiot, 2023-11-20:
Looks like ESPHome 2023.11.x version has something wrong with new BLE enable/disable changes.
Let's wait for next version like 2023.11.3 and if problem persist, I will take a close look.

myhomeiot, 2023-11-23:
@czesiu20 The origins of the problem this [fix](https://github.com/esphome/esphome/pull/5704). Please try [new](https://github.com/myhomeiot/esphome-components/releases/tag/1.0.10) release and let's me know.
PS: The new release will work for older ESPHome as well.

czesiu20, 2023-11-23:
Thank you for the fix. Ble Gateway is working great, but the client is not.
I had to remove the client, cos the device seems to be dead.

myhomeiot, 2023-11-23:
BLE Gateway works without any changes, the problem was in BLE Host and this problem kill ESP BLE stack.
I doesn't check it with `framework: arduino` but with `esp-idf` everything should works. `esp-idf` framework save a lot of memory so BLE works much stable. If you configuration allow to use `esp-idf` framework please try it and let's me know.

czesiu20, 2023-11-23:
Nope. Doesn't work...

orzechszek, 2025-01-07:
Hello, 
Any updates with myhomeiot_ble_client?

myhomeiot, 2025-01-08:
@orzechszek Version [1.0.10](https://github.com/myhomeiot/esphome-components/releases/tag/1.0.10) fixes this issue, version [1.0.11](https://github.com/myhomeiot/esphome-components/releases/tag/1.0.11) fixes problems with ESPHome 2024.11.x
Which error do you get, what is not working in your case?

orzechszek, 2025-01-08:
I don't see any updates in sensors.
Code with standard ble_client is working and sensors are updated in HA.

My code based on templates needed for client:

`
myhomeiot_ble_host:
myhomeiot_ble_client:
  - mac_address: XX:XX:XX:XX:XX:XX
    service_uuid: '180f'
    characteristic_uuid: '2a19'
    update_interval: 2min
    on_value:
      then:
        lambda: |-
          id(s_bat).publish_state(x[0]);

sensor:
  - platform: template
    name: "SBat"
    id: s_bat
    icon: 'mdi:battery'
    unit_of_measurement: '%'
`

Is there anything else what I miss?

myhomeiot, 2025-01-08:
I use it for HHCCJCY01 and it's works. Which type of framework do you use `framework: esp-idf` or arduino?
If you see very verbose logs (VV) you should see informational and debug messages from `myhomeiot_ble_client` if it's working and `MyHomeIOT BLE Client` in config log which appears first when you connect to ESP.

orzechszek, 2025-01-08:
I've just tested with esp-idf and it's also not working. In logs I don't see any notes with `myhomeiot_ble_client`.
I'm trying to connect with Oclean toothbrush.

myhomeiot, 2025-01-08:
It's can be because `ble_proxy` optimization, can you add regular `ble_client` with wrong MAC without lambda and try again?

orzechszek, 2025-01-09:
Something like will be OK? 

`ble_client:
  - mac_address: FF:FF:FF:FF:FF:FF
    id: test
    auto_connect: true`

myhomeiot, 2025-01-10:
> Something like below will be OK?

Yes

Do you use `bluetooth_proxy` in your configuration?

orzechszek, 2025-01-11:
I've tested and I don't see any logs with myhomeiot_ble_client. I dont use `bluetooth_proxy`

myhomeiot, 2025-01-12:
> I've tested and I don't see any logs with myhomeiot_ble_client. I dont use `bluetooth_proxy`

Very strange, can you remove all sensitive information from your config .yaml file and post it somewhere? I will try to understand what is wrong.

orzechszek, 2025-01-12:
My code

`substitutions:
  name: test
  friendly_name: test

external_components:
  - source: github://myhomeiot/esphome-components

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32dev
  framework:
    type: esp-idf

logger:
  level: VERY_VERBOSE

api:

ota:
- platform: esphome

improv_serial:

wifi:

esp32_ble_tracker:
  scan_parameters:
    interval: 200ms
    window: 80ms
    duration: 1min
    continuous: false

myhomeiot_ble_host:

myhomeiot_ble_client:
  - mac_address: XX:XX:XX:XX:XX:XX
    service_uuid: '180f'
    characteristic_uuid: '2a19'
    update_interval: 2min
    on_value:
      then:
        lambda: |-
          id(test1_bat).publish_state(x[0]);

  - mac_address: XX:XX:XX:XX:XX:XX
    service_uuid: '180f'
    characteristic_uuid: '2a19'
    update_interval: 2min
    on_value:
      then:
        lambda: |-
          id(test2_bat).publish_state(x[0]);

sensor:
  - platform: template
    name: "test1"
    id: test1_bat
    icon: 'mdi:battery'
    unit_of_measurement: '%'
  - platform: template
    name: "test2"
    id: test2_bat
    icon: 'mdi:battery'
    unit_of_measurement: '%'`

myhomeiot, 2025-01-13:
Why you use so strange `scan_parameters` try to comment it, or just change `continuous` to `true`.
`continuous: false` doesn't allow BT stack to listen.

myhomeiot, 2025-01-15:
@orzechszek Is its helps?

orzechszek, 2025-01-15:
Hello,

Sorry for late response. Finally I've tested it today, It seems it's working. Thanks!!
I need to verify it with more than 2 devices.
I have additional question - is there a possibility (some method?) to disable interval and trigger it manualy?
Something like: 
`
- myhomeiot_ble_client.update:
    id: my_ble_client

## or lambda

 - lambda: |-
      id(my_ble_client).update();
`

myhomeiot, 2025-01-15:
Great!
Yes, it's possible to trigger it manually by calling update method. You can also set `update_interval` to something huge or `never` (which equal to 4294967295).
Update method mark that value should be updated next time when device will be discovered by Bluetooth scanning, so it's can take some time.

orzechszek, 2025-01-15:
Is there any timeout? What if something appears once a day? 
I'm also getting error:
`
'virtual void esphome::myhomeiot_ble_client::MyHomeIOT_BLEClient::update()' is protected within this context
 id(device).update();
`

myhomeiot, 2025-01-15:
> Is there any timeout on update? What if something appears once a day?

If you sent `update_interval: 2h` it's will update immediately after ESP starts and will wait 2h when it's will try to update again, if device will not appear (for example device is fully discharged or out of the area) it's will wait until device will back and update it, if in some reason update fails, it's will trying to update next time when device become available.

> 'virtual void esphome::myhomeiot_ble_client::MyHomeIOT_BLEClient::update()' is protected within this context id(device).update();

I will check if it's possible to call update somehow.

orzechszek, 2025-01-15:
If I set update_interval more often, is it queued or cancelled if another request exists?

myhomeiot, 2025-01-15:
> If I set update_interval more often, is it queued or cancelled if another request exists?

It's doesn't have request queue it's just flag that info should be updated, if it's already set and waiting device to be available the next update just set this flag again to true but then device become available it's will update only once. This flag is reset only then device reply, if not it's will continue to trying to update.

orzechszek, 2025-01-20:
Hi, 

Any news regarding manual update? Is it possible?

myhomeiot, 2025-01-20:
> Any news regarding manual update? Is it possible?

Keep looking how to make it possible.

myhomeiot, 2025-01-24:
> Any news regarding manual update? Is it possible?

Sorry for delay, I thing the best way will be to move update method to public section, so it's can be called outside. The new release [1.0.12](https://github.com/myhomeiot/esphome-components/releases/tag/1.0.12) has this changes and now you can use it from lambda:
```yaml
button:
  - platform: template
    id: button_flower_care_update
    name: Flower Care Update
    on_press:
      - logger.log: "Button pressed"
      - lambda: |-
          id(bedrooms_flower_care).update();
```

myhomeiot, 2025-01-28:
@orzechszek New version works for you?

orzechszek, 2025-01-28:
I'm still testing my code, but your solution seems to work fine. Thanks!!

myhomeiot, 2025-01-29:
> I'm still testing my code, but your solution seems to work fine. Thanks!!

Great! Thanks for update.