myhomeiot/esphome-components #27: BLE client send command

chrisdev524, 2024-02-21:
Hi! 

It would be great to have a feature not only read values from BLE client devices, but sending commands. I have some BLE key finders and they have a feature to beep if I would like to find them. Now I can only trigger it via app. 

Could you extend the myhomeiot_ble_client with a functionality to connect to the device, write command, disconnect if a button is pressed like with ble_client.ble_write?

button:
  - platform: template
    name: "Nut alert"
    on_press:
      - ble_client.ble_write:
          id: nut
          service_uuid: "ff00"
          characteristic_uuid: "ff01"
          value: [0x03]
      - delay: 5s
      - ble_client.ble_write:
          id: nut
          service_uuid: "ff00"
          characteristic_uuid: "ff01"
          value: [0x04]

Thank you!

Archomeda, 2025-01-31:
I would also like this feature.

I like the idea of only connecting whenever necessary, because my Valve Index basestations (lighthouse v2) require themselves to be advertising all the time. I can already periodically poll successfully with this library, but I also want to manipulate the characteristic values by writing to them.
Currently that's not possible. And the official ble_client.connect is failing to work for me: esphome/issues#6443, so I'm kinda stuck with the implementation of my idea (able to poll the power state of the basestations, and also have the ability to turn them on and off).

happyiphone, 2025-02-11:
This would be great if it could be added.

myhomeiot, 2025-02-14:
@Archomeda @happyiphone @chrisdev524 Originally my BLE Client implementation was temporary solution until ESPHome devs will bring something better like Nimble BLE, but unfortunately we still have what we have :(

I found this [fork](https://github.com/dvb6666/esphome-components) which, I think, can help you. Please try it and let's me know.