myhomeiot/esphome-components #29: passive ble and esphome components sensors not send any values

ricain59, 2024-04-27:
Hi,

I've a sensor connected on passive ble, with a usb bluetooth connected on my home assistant and working fine various weeks ago.
I try connect this sensor on BLE gateway, but send the first time value and not send anymore any values.

Maybe i've put or done anything wrong and need your help.

It's my esp32 board: https://pt.aliexpress.com/item/1005006220389074.html?spm=a2g0o.order_list.order_list_main.45.3e1bcaa4BezpRe&gatewayAdapt=glo2bra

My esphome code on esp32:
```
substitutions:
  devicename: esp32-blegateway
  upper_devicename: esp32-blegateway

esphome:
  name: $devicename

external_components:
  - source: github://myhomeiot/esphome-components

esp32:
  board: esp32dev
  framework:
    type: arduino

mqtt:
  broker: x.x.x.x
  username: !secret esphomemqttlogin
  password: !secret esphomemqttpass
  port: 1883

wifi:
  ssid: !secret esphomewifi
  password: !secret esphomewifipass
  reboot_timeout: 5min
  fast_connect: on
  domain: !secret esphomedomain

logger:

api:
  encryption:
    key: !secret esphomeapiha   

ota:
  password: !secret esphomeota
  
web_server:
  port: 80

esp32_ble_tracker:

ble_gateway:
  id: blegateway
  on_ble_advertise:
    then:
      homeassistant.event:
        event: esphome.on_ble_advertise
        data:
          packet: !lambda return packet;
#          gateway_id: ${device_id}

binary_sensor:
  - platform: homeassistant
    id: ble_gateway_discovery
    entity_id: binary_sensor.ble_gateway
    attribute: discovery
    on_state:
      then:
        lambda: id(blegateway).set_discovery(x);

text_sensor:
  - platform: homeassistant
    id: ble_gateway_devices
    entity_id: binary_sensor.ble_gateway
    attribute: devices
    on_value:
      then:
        lambda: id(blegateway).set_devices(x);

switch:
  - platform: template
    id: switch_ble_gateway_discovery
    name: BLE Gateway Discovery
    icon: mdi:bluetooth-connect
    lambda: return id(blegateway).get_discovery();
    turn_on_action: [lambda: id(blegateway).set_discovery(true);]
    turn_off_action: [lambda: id(blegateway).set_discovery(false);]
    disabled_by_default: true
    entity_category: config

```

Put this on home assistant:
```
input_boolean:
  settings_ble_gateway:
    name: BLE Gateway
    icon: mdi:bluetooth
  settings_ble_gateway_discovery:
    name: BLE Gateway Discovery
    icon: mdi:bluetooth-connect

input_text:
  settings_ble_gateway_add_device:
    name: BLE Gateway Add Device
    icon: mdi:bluetooth-connect
    initial: ''

template:
  - binary_sensor:
      - name: BLE Gateway
        icon: mdi:bluetooth
        state: "{{ is_state('input_boolean.settings_ble_gateway', 'on') }}"
        attributes:
          discovery: "{{ is_state('input_boolean.settings_ble_gateway_discovery', 'on') }}"
          # devices: "{{ states | selectattr('entity_id', 'search', '^(device_tracker|sensor).ble_') | selectattr('attributes.mac address', 'defined') | map(attribute='attributes.mac address') | unique | sort | join('') | replace(':', '') ~ (states('input_text.settings_ble_gateway_add_device') | replace(':', '') | trim) if is_state('binary_sensor.ble_gateway', 'on') }}"
          # Important note: In Passive BLE Monitor version 7.8.2 and later 'attributes.mac address' was changed to 'attributes.mac_address', please update your config
          # devices: "{{ states | selectattr('entity_id', 'search', '^(device_tracker|sensor).ble_') | selectattr('attributes.mac_address', 'defined') | map(attribute='attributes.mac_address') | unique | sort | join('') | replace(':', '') ~ (states('input_text.settings_ble_gateway_add_device') | replace(':', '') | trim) if is_state('binary_sensor.ble_gateway', 'on') }}"
          # Note: In Home Assistant 2022.x, Passive BLE Monitor version 8.x and later you can use device attribute identifiers
          devices: >-
            {% set devices = namespace(items = []) %}
            {% for s in states | selectattr('entity_id', 'search', '^(device_tracker|sensor).ble_') | map(attribute='entity_id') %}
              {% set devices.items = devices.items + ([device_id(s)] if device_id(s) else []) %}
            {% endfor %}
            {% set ns = namespace(items = []) %}
            {% for s in devices.items | unique %}
              {% set ns.items = ns.items + [(device_attr(s, 'identifiers') | first)[1]] %}
            {% endfor %}
            {{ ns.items | unique | sort | join('') | replace(':', '') ~ (states('input_text.settings_ble_gateway_add_device') | replace(':', '') | trim) if is_state('binary_sensor.ble_gateway', 'on') }}
```
And create this automation on Home assistant:
```
- id: ESPHome BLE Advertise
  alias: ESPHome BLE Advertise
  mode: queued
  trigger:
    - platform: event
      event_type: esphome.on_ble_advertise
  action:
    - service: ble_monitor.parse_data
      data:
        packet: "{{ trigger.event.data.packet }}"
```

What ti's wrong on my configuration?

Thank you for help.

myhomeiot, 2024-04-27:
Hello,

Do you see the packets in ESPHome log? This log should looks like below, but be sure that you get messages from your device by MAC address, in this example it's `2A:03:4E:CF:0F:38`.
```
[06:06:47][D][ble_gateway:063]: [2A:03:4E:CF:0F:38] Packet 043E2D02010000BA....
```

If your device requires active scanning, you should add this parameter into esp32_ble_tracker section (more info you can find [here](https://esphome.io/components/esp32_ble_tracker.html)):
```yaml
esp32_ble_tracker:
  scan_parameters:
    active: true
```

ricain59, 2024-04-27:
Hello,

> Do you see the packets in ESPHome log? This log should looks like below, but be sure that you get messages from your device by MAC address, in this example it's `2A:03:4E:CF:0F:38`.
> 
> ```
> [06:06:47][D][ble_gateway:063]: [2A:03:4E:CF:0F:38] Packet 043E2D02010000BA....
> ```
Yes, but only the first time when i active input_boolean.settings_ble_gateway_discovery on home assistant. And after that, never more the sensor send data to the home assistant and on esphome log never appear again this log.

> If your device requires active scanning, you should add this parameter into esp32_ble_tracker section (more info you can find [here](https://esphome.io/components/esp32_ble_tracker.html)):
> 
> ```yaml
> esp32_ble_tracker:
>   scan_parameters:
>     active: true
> ```

I go to try this configuration.
Thank you for your time.

myhomeiot, 2024-04-27:
> Yes, but only the first time when i active input_boolean.settings_ble_gateway_discovery on home assistant. And after that, never more the sensor send data to the home assistant and on esphome log never appear again this log.

`input_boolean.settings_ble_gateway_discovery` should allow to send all BLE packets into Home Assistant, not just from your device, so you should see many packets in the log sure if you have other BLE devices.

myhomeiot, 2024-04-29:
You can also try to use esp-idf framework, bluetooth in this framework works better.

```yaml
esp32:
  board: esp32dev
  framework:
    type: esp-idf
```

So to make things simpler use this minimal config in your ESPHome device, where I mark important changes:
```yaml
substitutions:
  devicename: esp32-blegateway
  upper_devicename: esp32-blegateway

esphome:
  name: $devicename

external_components:
  - source: github://myhomeiot/esphome-components

esp32:
  board: esp32dev
  framework:
    type: esp-idf  # <--------------------------------

mqtt:
  broker: x.x.x.x
  username: !secret esphomemqttlogin
  password: !secret esphomemqttpass
  port: 1883

wifi:
  ssid: !secret esphomewifi
  password: !secret esphomewifipass
  reboot_timeout: 5min
  fast_connect: on
  domain: !secret esphomedomain

logger:

api:
  encryption:
    key: !secret esphomeapiha   

ota:
  password: !secret esphomeota
  
web_server:
  port: 80

esp32_ble_tracker:
  scan_parameters:
    active: true  # <--------------------------------

ble_gateway:
  id: blegateway
  discovery: true  # <--------------------------------
  on_ble_advertise:
    then:
      homeassistant.event:
        event: esphome.on_ble_advertise
        data:
          packet: !lambda return packet;
#          gateway_id: ${device_id}
```

If this doesn't help when I need more information about your device and maybe some logs from Passive BLE Monitor when device connected by USB dongle.

ricain59, 2024-05-01:
Hello,
I've done this changes 1 hours ago and working fine. 
I'm waiting more days and give more feedback.

Thank you for your help and your time.

ricain59, 2024-05-05:
Hello,
It's working fine.

Thank you for your help.

closed 2024-05-05 by ricain59