myhomeiot/esphome-components #30: WARNING Can't connect to ESPHome API for screek-ble-monitor-1a.local: Error resolving IP address:

dbrb2, 2024-06-28:
Hi, 
I've been trying to set up an ESP32 classic with your code, so that I can void using directly connected USB Bluetooth dongles. 
I have a Home Assistant instance, and the ESP32 yaml below - but am getting the API error shown at the bottom of this post....are you able to offer any suggestions as to what I am doing wrong? 

```
api:
  encryption:
    key: xxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esphome:
  name: screek-ble-monitor-1a
  friendly_name: BLE Monitor Gateway 1A
  name_add_mac_suffix: True
  project: 
    name: Screek.BLE Monitor Gateway
    version: 1A

external_components:
  - source: github://myhomeiot/esphome-components

logger:

esp32:
  variant: ESP32
  board: esp32dev
  framework:
    type: arduino 
    version: 2.0.7
    platform_version: 6.0.1

captive_portal:

improv_serial:

esp32_ble_tracker:

sensor:
  - platform: uptime
    name: Uptime
    id: sys_uptime
    update_interval: 10s
  - platform: wifi_signal 
    name: RSSI
    id: wifi_rssi
    update_interval: 10s
    entity_category: "diagnostic"
  - platform: template
    id: esp_memory
    icon: mdi:memory
    name: ESP Free Memory
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL) / 1024;
    unit_of_measurement: 'kB'
    state_class: measurement
    entity_category: "diagnostic"
    
ble_gateway:
  id: blegateway
  on_ble_advertise:
    then:
      homeassistant.event:
        event: esphome.on_ble_advertise
        data:
          packet: !lambda return packet;

binary_sensor:
  - platform: homeassistant
    id: ble_gateway_discovery
    entity_id: binary_sensor.ble_gateway
    attribute: discovery
    on_state:
      then:
        lambda: id(blegateway).set_discovery(x);
  - platform: status
    name: Online
    id: ink_ha_connected

text_sensor:
  - platform: homeassistant
    id: ble_gateway_devices
    entity_id: binary_sensor.ble_gateway
    attribute: devices
    on_value:
      then:
        lambda: id(blegateway).set_devices(x);

switch:
  - platform: template
    id: switch_ble_gateway_discovery
    name: BLE Gateway Discovery
    icon: mdi:bluetooth-connect
    lambda: return id(blegateway).get_discovery();
    turn_on_action: [lambda: id(blegateway).set_discovery(true);]
    turn_off_action: [lambda: id(blegateway).set_discovery(false);]
    disabled_by_default: true
    entity_category: config

button:
  - platform: restart
    # disabled_by_default: True
    icon: mdi:power-cycle
    name: "ESP Reboot"
  - platform: factory_reset
    disabled_by_default: True
    name: Factory Reset
    id: factory_reset_al
```

```
INFO ESPHome 2024.6.3
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-0f0bd4.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Starting log output from screek-ble-monitor-1a.local using esphome API
WARNING Can't connect to ESPHome API for screek-ble-monitor-1a.local: Error resolving IP address: [Errno -5] No address associated with hostname (APIConnectionError)
INFO Trying to connect to screek-ble-monitor-1a.local in the background
```

dbrb2, 2024-06-29:
Ah - the issue seems to have been resolved by adding a specific IP address to the wifi YAML section

myhomeiot, 2024-06-30:
> Ah - the issue seems to have been resolved by adding a specific IP address to the wifi YAML section

Now it's works? The component can't be source of this problem, you can try to remove only `ble_gateway` part and see how this works without component.

dbrb2, 2024-06-30:
Yes, the problem was that mDNS was not working on my network.

By specifying the io address in the esp32 yaml, this problem was avoided

closed 2024-07-01 by myhomeiot