myhomeiot/esphome-components #32: Cannot compile with ESPHome 2024.11.0

rossowl, 2024-11-21:
Compilation error:

src/main.cpp: In function 'void setup()':
src/main.cpp:389:87: error: invalid new-expression of abstract class type 'esphome::myhomeiot_ble_host::MyHomeIOT_BLEHost'
   myhomeiot_ble_host_myhomeiot_blehost_id = new myhomeiot_ble_host::MyHomeIOT_BLEHost();
                                                                                       ^
In file included from src/esphome/components/myhomeiot_ble_client/myhomeiot_ble_client.h:8,
                 from src/esphome.h:71,
                 from src/main.cpp:3:
src/esphome/components/myhomeiot_ble_host/myhomeiot_ble_host.h:47:7: note:   because the following virtual functions are pure within 'esphome::myhomeiot_ble_host::MyHomeIOT_BLEHost':
 class MyHomeIOT_BLEHost : public Component, public esp32_ble_tracker::ESPBTClient {
       ^~~~~~~~~~~~~~~~~
In file included from src/esphome/components/ble_gateway/ble_gateway.h:6,
                 from src/esphome.h:18,
                 from src/main.cpp:3:
src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.h:175:16: note: 	'virtual void esphome::esp32_ble_tracker::ESPBTClient::disconnect()'
   virtual void disconnect() = 0;

myhomeiot, 2024-11-21:
Fixed in [1.0.11](https://github.com/myhomeiot/esphome-components/releases/tag/1.0.11), please check.

rossowl, 2024-11-21:
Hi,
Thank you for fast response, it is not important error right now.
Hass.io addon, click Clean build files, Update...
Still same error. Maybe I dont know how clean src folder and where is .pioenvs.
Easy.
I will try it again later.

INFO ESPHome 2024.11.0
INFO Reading configuration /config/esphome/esp32-koupelna.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-koupelna (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/esp32-koupelna/src/esphome/components/api/api_connection.o
...
...
...
Compiling .pioenvs/esp32-koupelna/bt/common/btc/core/btc_alarm.o
src/main.cpp: In function 'void setup()':
src/main.cpp:357:87: error: invalid new-expression of abstract class type 'esphome::myhomeiot_ble_host::MyHomeIOT_BLEHost'
   myhomeiot_ble_host_myhomeiot_blehost_id = new myhomeiot_ble_host::MyHomeIOT_BLEHost();
                                                                                       ^
In file included from src/esphome/components/myhomeiot_ble_client/myhomeiot_ble_client.h:8,
                 from src/esphome.h:71,
                 from src/main.cpp:3:
src/esphome/components/myhomeiot_ble_host/myhomeiot_ble_host.h:47:7: note:   because the following virtual functions are pure within 'esphome::myhomeiot_ble_host::MyHomeIOT_BLEHost':
 class MyHomeIOT_BLEHost : public Component, public esp32_ble_tracker::ESPBTClient {
       ^~~~~~~~~~~~~~~~~
In file included from src/esphome/components/ble_gateway/ble_gateway.h:6,
                 from src/esphome.h:18,
                 from src/main.cpp:3:
src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.h:175:16: note: 	'virtual void esphome::esp32_ble_tracker::ESPBTClient::disconnect()'
   virtual void disconnect() = 0;
                ^~~~~~~~~~
Compiling .pioenvs/esp32-koupelna/bt/common/api/esp_blufi_api.o
Compiling .pioenvs/esp32-koupelna/bt/common/hci_log/bt_hci_log.o
Compiling .pioenvs/esp32-koupelna/bt/common/btc/core/btc_manage.o
Compiling .pioenvs/esp32-koupelna/bt/common/btc/core/btc_task.o
Compiling .pioenvs/esp32-koupelna/bt/common/btc/profile/esp/blufi/blufi_prf.o
Compiling .pioenvs/esp32-koupelna/bt/common/btc/profile/esp/blufi/blufi_protocol.o
Compiling .pioenvs/esp32-koupelna/bt/common/osi/alarm.o
Compiling .pioenvs/esp32-koupelna/bt/common/osi/allocator.o
Compiling .pioenvs/esp32-koupelna/bt/common/osi/buffer.o
Compiling .pioenvs/esp32-koupelna/bt/common/osi/config.o
Compiling .pioenvs/esp32-koupelna/bt/common/osi/fixed_queue.o
Compiling .pioenvs/esp32-koupelna/bt/common/osi/pkt_queue.o
Compiling .pioenvs/esp32-koupelna/bt/common/osi/fixed_pkt_queue.o
Compiling .pioenvs/esp32-koupelna/bt/common/osi/future.o
Compiling .pioenvs/esp32-koupelna/bt/common/osi/hash_functions.o
*** [.pioenvs/esp32-koupelna/src/main.o] Error 1
========================= [FAILED] Took 66.62

rossowl, 2024-11-22:
ESPHome upgrade to 2024.11.1
Clean build files
Install
======================== [SUCCESS] Took 220.30 seconds ========================
Thank you!

closed 2024-11-22 by rossowl