myhomeiot/esphome-components #33: HolyIOT devices are only partially detected with Gateway + Passive BLE Monitor

WackEarl, 2024-11-27:
Hi there,

first of thanks for this awesome project :)

I have a little issue where HolyIOT devices are only partially recognized in BLE Monitor when packages are forwarded by the gateway. They work correctly if in range of the server running BLE Monitor with it's built-in bluetooth chip.
These devices are kind of special I guess, as BLE Monitor will create 2 entries for each. (see here: https://github.com/custom-components/ble_monitor/issues/1183#issuecomment-1568144854)
One is an iBeacon (Manufacturer Apple, identified by UUID), which is working with the gateway as well. The other one is actually Manufacturer HolyIOT with additional sensors (e.g. vibration), identified by MAC. This is not working with the gateway.
I'm assuming it does additional iBeacon broadcasts to the HolyIOT ones, which are converted(?) correctly whereas the later ones aren't.

My setup is really simple, forwarding everything with direct parse service call (no events). Discovery is enabled for BLE Monitor and Gateway.
Again: iBeacon is created/detected for both, but HolyIOT only if in proximity to server - and lost if I move it away to a gateway.

Maybe sth. is lost when converting the packages? Or am I missing sth. here why this should/can not work at all?

Thank you :)

myhomeiot, 2024-11-27:
Hello,

First I'm glad that you happy with this project ;-) It's always nice to hear.
You can try `active: true` for `esp32_ble_tracker` sometimes it's helps:
```yaml
esp32_ble_tracker:
  scan_parameters:
    active: true  # <--------------------------------
```

Some info you can also find in this [issue](https://github.com/myhomeiot/esphome-components/issues/29).

WackEarl, 2024-11-27:
Now I feel really dumb. That was actually it. 
I ignored this as issue because BLE Monitor was running with passive only as well.

Well whatever. Solved. 
Thank you so much for your quick reply and have a nice day!

closed 2024-11-27 by WackEarl