myhomeiot/esphome-components #34: Not sure where to start

johntdyer, 2025-02-14:
I have dozens of BLE devices around my home and I've deployed BLE proxies using GL-s10's

my home assistant config section

```
ble_monitor:
  bt_interface: "disable"

automation:

  - alias: ESPHome BLE Advertise 2
    id: 6d12c529-fa6c-4a4e-ba30-df4779b5ddb9
    mode: queued
    trigger:
      - platform: event
        event_type: esphome.on_ble_advertise
    action:
      - service: ble_monitor.parse_data
        data:
          packet: "{{ trigger.event.data.packet }}"
          gateway_id: "{{ trigger.event.data.gateway_id if trigger.event.data.gateway_id is defined else 'unknown' }}"

input_boolean:
  settings_ble_gateway:
    name: BLE Gateway
    icon: mdi:bluetooth
  settings_ble_gateway_discovery:
    name: BLE Gateway Discovery
    icon: mdi:bluetooth-connect

input_text:
  settings_ble_gateway_add_device:
    name: BLE Gateway Add Device
    icon: mdi:bluetooth-connect
    initial: ''

template:
  - binary_sensor:
      - name: BLE Gateway
        icon: mdi:bluetooth
        state: "{{ is_state('input_boolean.settings_ble_gateway', 'on') }}"
        attributes:
          discovery: "{{ is_state('input_boolean.settings_ble_gateway_discovery', 'on') }}"
          # devices: "{{ states | selectattr('entity_id', 'search', '^(device_tracker|sensor).ble_') | selectattr('attributes.mac address', 'defined') | map(attribute='attributes.mac address') | unique | sort | join('') | replace(':', '') ~ (states('input_text.settings_ble_gateway_add_device') | replace(':', '') | trim) if is_state('binary_sensor.ble_gateway', 'on') }}"
          # Important note: In Passive BLE Monitor version 7.8.2 and later 'attributes.mac address' was changed to 'attributes.mac_address', please update your config
          # devices: "{{ states | selectattr('entity_id', 'search', '^(device_tracker|sensor).ble_') | selectattr('attributes.mac_address', 'defined') | map(attribute='attributes.mac_address') | unique | sort | join('') | replace(':', '') ~ (states('input_text.settings_ble_gateway_add_device') | replace(':', '') | trim) if is_state('binary_sensor.ble_gateway', 'on') }}"
          # Note: In Home Assistant 2022.x, Passive BLE Monitor version 8.x and later you can use device attribute identifiers
          devices: >-
            {% set devices = namespace(items = []) %}
            {% for s in states | selectattr('entity_id', 'search', '^(device_tracker|sensor).ble_') | map(attribute='entity_id') %}
              {% set devices.items = devices.items + [device_id(s)] %}
            {% endfor %}
            {% set ns = namespace(items = []) %}
            {% for s in devices.items | unique %}
              {% set ns.items = ns.items + [(device_attr(s, 'identifiers') | first)[1]] %}
            {% endfor %}
            {{ ns.items | unique | sort | join('') | replace(':', '') ~ (states('input_text.settings_ble_gateway_add_device') | replace(':', '') | trim) if is_state('binary_sensor.ble_gateway', 'on') }}

```

my ESP32-S3 lite config in esphome

```
esphome:
  name: ble-gateway
  friendly_name: ble-gateway

external_components:
  - source: github://myhomeiot/esphome-components

esp32:

  board: esp32-s3-devkitc-1

  framework:
    type: arduino

# Enable logging

logger:
  # level: DEBUG
  level: DEBUG
  # VERY_VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: "192.168.121.47"
  # Enable fallback hotspot (captive portal) in case wifi connection fails

captive_portal:

improv_serial:

sensor:
  - platform: uptime
    name: Uptime
    id: sys_uptime
    update_interval: 10s
  - platform: wifi_signal
    name: RSSI
    id: wifi_rssi
    update_interval: 10s
    entity_category: "diagnostic"
  - platform: template
    id: esp_memory
    icon: mdi:memory
    name: ESP Free Memory
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL) / 1024;
    unit_of_measurement: 'kB'
    state_class: measurement
    entity_category: "diagnostic"

binary_sensor:
  - platform: homeassistant
    id: ble_gateway_discovery
    entity_id: binary_sensor.ble_gateway
    attribute: discovery
    on_state:
      then:
        lambda: id(blegateway).set_discovery(x);

  - platform: status
    name: Online
    id: ink_ha_connected

text_sensor:
  - platform: homeassistant
    id: ble_gateway_devices
    entity_id: binary_sensor.ble_gateway
    attribute: devices
    on_value:
      then:
        lambda: id(blegateway).set_devices(x);

switch:
  - platform: template
    id: switch_ble_gateway_discovery
    name: BLE Gateway Discovery
    icon: mdi:bluetooth-connect
    lambda: return id(blegateway).get_discovery();
    turn_on_action: [lambda: id(blegateway).set_discovery(true);]
    turn_off_action: [lambda: id(blegateway).set_discovery(false);]
    disabled_by_default: true
    entity_category: config

button:
  - platform: restart
    # disabled_by_default: True
    icon: mdi:power-cycle
    name: "ESP Reboot"
  - platform: factory_reset
    disabled_by_default: True
    name: Factory Reset
    id: factory_reset_all

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - homeassistant.event:
          event: esphome.on_ble_advertise
          data:
            packet: !lambda return ble_gateway::BLEGateway::scan_result_to_hci_packet_hex(x.get_scan_result());
#            gateway_id: ${device_id}
            proxy: !lambda return true;
      # - lambda: |-
      #    ESP_LOGI("on_ble_advertise", "[%s] Packet %s", x.address_str().c_str(),
      #      ble_gateway::BLEGateway::scan_result_to_hci_packet_hex(x.get_scan_result()).c_str());

ble_gateway:
  id: blegateway
  discovery: True

  on_ble_advertise:
    then:
      homeassistant.event:
        event: esphome.on_ble_advertise
        data:
          packet: !lambda return packet;

```

I turn on the input_boolean switches and the logs of the ESP 32 roll by with tons of items but nothing descriptive about any of them,  just a sea of payload with nothing to tell me what that device is

Example:

logs_ble-gateway_logs.txt

I tried to check the automation traces but there is nothing descriptive there, in fact it seemingly DDOS's my machine if I used `queued` mode :) 

so I guess I am at a loss on where to start , I would love to start instrumenting some of my bluetooth devices, like my toothbrush and coffee maker but I have no clue how to tell what they are in this sea of noise.  Appreciate any help you can provide

johntdyer, 2025-02-14:
[logs_ble-gateway_logs.txt](https://github.com/user-attachments/files/18804681/logs_ble-gateway_logs.txt)

myhomeiot, 2025-02-14:
Hello, not sure what you want to achieve, but your configuration looks more or less correct and after you turn on `BLE Gateway Discovery` tons of items in the log it's BLE Advertisement packets which goes to Home Assistant and forwards to `Passive BLE Monitor` by `ESPHome BLE Advertise 2` automation. As you have `Passive BLE Monitor` [discovery](https://custom-components.github.io/ble_monitor/configuration_params#discovery) enabled by default after some time you should see all discovered devices with data in Passive BLE Monitor or some messages about problems in the Home Assistant log.

If you have many devices nearby, instead enabling discovery probably will be better to add your devices MAC one by one specify (or all together without spaces) in `input_text.settings_ble_gateway_add_device` in this case only packets from this devices will be forwarded to Home Assistant.

You have `on_ble_advertise` on `esp32_ble_tracker` and `ble_gateway` sections, if you want to use MAC address filtering - remove `on_ble_advertise` from `esp32_ble_tracker` because it's forwards all BLE packets into Home Assistant. Recommended configuration example you can see [here](https://github.com/myhomeiot/esphome-components/blob/main/examples/ble_gateway/esphome.yaml) for ESPHome and [here](https://github.com/myhomeiot/esphome-components/blob/main/examples/ble_gateway/homeassistant.yaml) for Home Assistant.
```yaml
esp32_ble_tracker:
  on_ble_advertise: # <--- Forward all packets to Home Assistant, remove this
  ...

ble_gateway:
  id: blegateway
  discovery: True # <--- Forward all packets to Home Assistant, remove it and turn on when needed
  on_ble_advertise:
  ...
```

johntdyer, 2025-02-15:
Thank you very much, honestly I am not sure entirely what I want to achieve either but I am just experimenting..  I want to see what devices are near by in my home and what data they are advertising and what I can do with that data.  How do I find the mac address of any given component to even filter upon it ?

johntdyer, 2025-02-15:
I downloaded app called BT inspector for example and brought my Sonicare toothbrush near by and I got something along the lines of the following

[image]

but its not clear to me how I add this and start experimenting with it.   Given the multitude of ble devices this day how does one filter through all the noise the find any particular device in general and then start exploring it to add into this tool chain ?  Sorry if these are naive questions, just very new to the BLE ecosystem

myhomeiot, 2025-02-16:
Passive BLE Monitor (with BLE Gateway as remote adapter) support only BLE Advertisement and not for all BLE devices, full list of supported devices you can find [here](https://custom-components.github.io/ble_monitor/devices)
You should understand difference between BLE Advertisement and BLE/Bluetooth communication, some devices send data using BLE Advertisement broadcast packages which can be captured and parsed by Passive BLE Monitor. If device requires connection the Passive BLE Monitor can't work with such devices and you should use regular ESPHome BLE Client or [Bluetooth Proxy from Home Assistant](https://www.home-assistant.io/integrations/bluetooth/#remote-adapters-bluetooth-proxies) which can make connection.

To get battery % using default Bluetooth service you can use this [example](https://github.com/myhomeiot/esphome-components/blob/main/examples/myhomeiot_ble_client/ble_battery.yaml) with BLE Client which make connection.

If your device support BLE Advertisement but not supported by Passive BLE Monitor then you can try to ask dev to add support by creating the issue [here](https://github.com/custom-components/ble_monitor/issues) and provide all required data.

Hope this helps.