myhomeiot/esphome-components #36: Error compiling on ESPHome 2025.6.0

lazyandproud, 2025-06-18:
INFO ESPHome 2025.6.0
INFO Reading configuration /config/esphome/ble-tracker.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ble-tracker (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/ble-tracker/src/esphome/components/ble_gateway/ble_gateway.cpp.o
Compiling .pioenvs/ble-tracker/src/esphome/components/esp32_ble/ble_advertising.cpp.o
Compiling .pioenvs/ble-tracker/src/esphome/components/esp32_ble/ble_uuid.cpp.o
Compiling .pioenvs/ble-tracker/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp.o
src/esphome/components/ble_gateway/ble_gateway.cpp: In member function 'virtual bool esphome::ble_gateway::BLEGateway::parse_device(const esphome::esp32_ble_tracker::ESPBTDevice&)':
src/esphome/components/ble_gateway/ble_gateway.cpp:62:56: error: 'const class esphome::esp32_ble_tracker::ESPBTDevice' has no member named 'get_scan_result'
     auto packet = scan_result_to_hci_packet_hex(device.get_scan_result());
                                                        ^~~~~~~~~~~~~~~
*** [.pioenvs/ble-tracker/src/esphome/components/ble_gateway/ble_gateway.cpp.o] Error 1
========================== [FAILED] Took 3.04 seconds ==========================

myhomeiot, 2025-06-19:
Fixed in version [1.0.13](https://github.com/myhomeiot/esphome-components/releases/tag/1.0.13) but required https://github.com/esphome/esphome/pull/9148, if you  don't want to wait this PR to be merged in order to apply this PR you can use following ESPHome configuration:
```yaml
external_components:
  - source: github://myhomeiot/esphome-components
  - source: github://pr#9148
    components: [esp32_ble_tracker]
```

**Don't forget to remove this configuration after this PR will be merged!**

mccaffm, 2025-06-19:
(inkyenv) pi@piw1:~/inky/examples $
(inkyenv) pi@piw1:~/inky/examples $ python3 name-badge.py --colour red --name "Test Badge"
Traceback (most recent call last):
  File "/home/pi/inky/examples/name-badge.py", line 7, in <module>
    from font_intuitive import Intuitive
ModuleNotFoundError: No module named 'font_intuitive'

mccaffm, 2025-06-19:
(inkyenv) pi@piw1:~/inky/examples $ python3 name-badge.py --colour red --name "Test Badge"
Traceback (most recent call last):
  File "/home/pi/inky/examples/name-badge.py", line 10, in <module>
    from inky.auto import auto
ModuleNotFoundError: No module named 'inky'