myhomeiot/esphome-components #39: ESPHome 2025.7.2 Unable to Update

weapons, 2025-07-20:
Hi.
I am unable to update to ESPHome 2025.7.2.  Getting the following error:

INFO ESPHome 2025.7.2
INFO Reading configuration /config/esphome/esphome-web-354a1c.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing screek-ble-monitor-1a (board: esp32dev; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 3.1.3 
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Dependency Graph
|-- Networking @ 3.1.3
|-- WiFi @ 3.1.3
|-- ESPmDNS @ 3.1.3
|-- Update @ 3.1.3
|-- Improv @ 1.2.4
Compiling .pioenvs/screek-ble-monitor-1a/src/esphome/components/ble_gateway/ble_gateway.cpp.o
Compiling .pioenvs/screek-ble-monitor-1a/src/esphome/components/factory_reset/button/factory_reset_button.cpp.o
Compiling .pioenvs/screek-ble-monitor-1a/src/esphome/components/homeassistant/binary_sensor/homeassistant_binary_sensor.cpp.o
Compiling .pioenvs/screek-ble-monitor-1a/src/esphome/components/homeassistant/text_sensor/homeassistant_text_sensor.cpp.o
In file included from src/esphome/components/ble_gateway/ble_gateway.cpp:3:
src/esphome/components/ble_gateway/ble_gateway.h:25:58: error: 'esp32_ble' does not name a type
   25 |   static std::string scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result);
      |                                                          ^~~~~~~~~
src/esphome/components/ble_gateway/ble_gateway.h:25:83: error: expected unqualified-id before '&' token
   25 |   static std::string scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result);
      |                                                                                   ^
src/esphome/components/ble_gateway/ble_gateway.h:25:82: error: expected ')' before '&' token
   25 |   static std::string scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result);
      |                                                   ~                              ^~
      |                                                                                  )
src/esphome/components/ble_gateway/ble_gateway.h:25:83: error: static member function 'static std::string esphome::ble_gateway::BLEGateway::scan_result_to_hci_packet_hex(...)' cannot have ref-qualifier
   25 |   static std::string scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result);
      |                                                                                   ^
src/esphome/components/ble_gateway/ble_gateway.h:25:83: error: expected ';' at end of member declaration
   25 |   static std::string scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result);
      |                                                                                   ^
      |                                                                                    ;
src/esphome/components/ble_gateway/ble_gateway.h:25:84: error: 'scan_result' does not name a type
   25 |   static std::string scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result);
      |                                                                                    ^~~~~~~~~~~
src/esphome/components/ble_gateway/ble_gateway.cpp:15:61: error: 'esp32_ble' does not name a type
   15 | std::string BLEGateway::scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result) {
      |                                                             ^~~~~~~~~
src/esphome/components/ble_gateway/ble_gateway.cpp:15:86: error: expected unqualified-id before '&' token
   15 | std::string BLEGateway::scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result) {
      |                                                                                      ^
src/esphome/components/ble_gateway/ble_gateway.cpp:15:85: error: expected ')' before '&' token
   15 | std::string BLEGateway::scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result) {
      |                                                      ~                              ^~
      |                                                                                     )
src/esphome/components/ble_gateway/ble_gateway.cpp:15:87: error: expected initializer before 'scan_result'
   15 | std::string BLEGateway::scan_result_to_hci_packet_hex(const esp32_ble::BLEScanResult &scan_result) {
      |                                                                                       ^~~~~~~~~~~
*** [.pioenvs/screek-ble-monitor-1a/src/esphome/components/ble_gateway/ble_gateway.cpp.o] Error 1
========================= [FAILED] Took 40.27 seconds =========================

weapons, 2025-07-21:
Hi.  Thanks for your help.  I'd already tried clearing the build files but no luck.  I guess I'll leave at 2025.6.3. Regards.

myhomeiot, 2025-07-21:
You can try to install 2025.6.3, when update to 2025.7.2 compile and attach here the full log from first compile (during first compile it's download necessary requirements) maybe it's will help me to help you.

myhomeiot, 2025-07-21:
I test with standalone ESPHome (regular PC installed by `pip3 install esphome==2025.7.2`) and also everything fine :(
Your problem with type `esp32_ble::BLEScanResult` which included in ESPHome 2025.7, maybe you use something like `source: github://pr#9148` for esp32_ble_tracker?
```yaml
external_components:
  - source: github://myhomeiot/esphome-components
  - source: github://pr#9148
    components: [esp32_ble_tracker]
```

First run usually looks like (you should see downloading and installing dependencies):

```
INFO ESPHome 2025.7.2
INFO Reading configuration opentherm.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing opentherm (board: esp32dev; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/espressif/arduino-esp32/releases/download/3.1.3/esp32-3.1.3.zip
INFO Installing https://github.com/espressif/arduino-esp32/releases/download/3.1.3/esp32-3.1.3.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32@3.1.3 has been installed!
INFO framework-arduinoespressif32@3.1.3 has been installed!
Tool Manager: Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.3/esp32-arduino-libs-idf-release_v5.3-489d7a2b-v1.zip
INFO Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.3/esp32-arduino-libs-idf-release_v5.3-489d7a2b-v1.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32-libs@5.3.0+sha.489d7a2b3a has been installed!
INFO framework-arduinoespressif32-libs@5.3.0+sha.489d7a2b3a has been installed!
Tool Manager: Installing platformio/toolchain-xtensa-esp-elf @ 13.2.0+20240530
INFO Installing platformio/toolchain-xtensa-esp-elf @ 13.2.0+20240530
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa-esp-elf@13.2.0+20240530 has been installed!
INFO toolchain-xtensa-esp-elf@13.2.0+20240530 has been installed!
Tool Manager: Installing platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
INFO Installing platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-xtensa-esp-elf-gdb@14.2.0+20240403 has been installed!
INFO tool-xtensa-esp-elf-gdb@14.2.0+20240403 has been installed!
Tool Manager: Installing platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
INFO Installing platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-riscv32-esp-elf-gdb@14.2.0+20240403 has been installed!
INFO tool-riscv32-esp-elf-gdb@14.2.0+20240403 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esptool/releases/download/v4.8.6/esptool.zip
INFO Installing https://github.com/pioarduino/esptool/releases/download/v4.8.6/esptool.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@4.8.6 has been installed!
INFO tool-esptoolpy@4.8.6 has been installed!
Tool Manager: Installing tasmota/tool-mklittlefs @ ^3.2.0
INFO Installing tasmota/tool-mklittlefs @ ^3.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-mklittlefs@3.2.0 has been installed!
INFO tool-mklittlefs@3.2.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (53.3.13) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.1.3
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a
 - tool-esptoolpy @ 4.8.6
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Installing Arduino Python dependencies
Collecting wheel>=0.35.1
  Using cached wheel-0.45.1-py3-none-any.whl.metadata (2.3 kB)
Using cached wheel-0.45.1-py3-none-any.whl (72 kB)
Installing collected packages: wheel
Successfully installed wheel-0.45.1
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Networking @ 3.1.3
|-- WiFi @ 3.1.3
|-- ESPmDNS @ 3.1.3
|-- Update @ 3.1.3
Building in release mode
Compiling .pioenvs\opentherm\src\esphome\components\api\api_connection.cpp.o
...
```

myhomeiot, 2025-07-23:
@weapons Try to update to ESPHome 2025.7.3 and compile. Update doesn't include anything for you problem but maybe fresh update will help, if no, I suggest you to reinstall ESPHome Device Builder/ESPHome.

weapons, 2025-07-23:
Hi. I removed the  - source: github://pr#9148
Line from yaml and updated with 2025.7.3 and all back to normal. It updated without any issues. 
Thank you very much for your time and help. 
Really appreciate.